LazeMSS/OctoPrint-UICustomizer #2: Layout Bug

JFI611, 2020-11-30:
I changed my layout to have "State" as the first option in my layout/view and hit save. Once I saved everything was fine. However, once I reloaded Octoprint, the "State" always moves as the last option in the list/view. I tested this on all three columns with the same result.

Guilouz, 2020-11-30:
Same here, change and save have no effect. I use last build.

LazeMSS, 2020-11-30:
Fixed in 0.0.9.4: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.0.9.4

LazeMSS, 2020-11-30:
> Same here, change and save have no effect. I use last build.
> 
> And plugins icons are centered, it must be on the right
> 
> [image]

The icons are centered by design when running on big screens. Please report this a seperate issue if you want it customizable

closed 2020-11-30 by LazeMSS

LazeMSS, 2020-12-05:
Next version will have this an option