LazeMSS/OctoPrint-UICustomizer #4: All widgets except Main tabs and Webcam disappear after reordering tabs using Dragon Order

lokety, 2020-11-30:
I've had https://plugins.octoprint.org/plugins/dragon_order/ installed before installing UI Customizer.

Now whenever I reorder the tabs, all three columns and their widgets disappear, leaving just the header and footer. Then, in the Layout section, Main tabs and Webcam appear under Left row (or should it be called column?) and nothing else.

Clearing data helps to restore all the widgets.

Perhaps a conflict between these two plugins?

NovaViper, 2020-12-01:
Same here! But for me, Tasmota seems to cause the same issue

LazeMSS, 2020-12-01:
I will look into but in general these plugins and mine collide because they basically manipulate with the same content inside the viewport. For now you will probally have to disable one of them.

NovaViper, 2020-12-01:
Tasmota oddly enough doesn't actually change up the viewpoint, not as drastically as Dragon does (it just adds
 a button in the nav bar and a new tab for powering monitoring)

NovaViper, 2020-12-01:
And even worse, the issue still happens even though I have Tasmota turned off.. so im not sure what it could be anymore <_>

LazeMSS, 2020-12-02:
> And even worse, the issue still happens even though I have Tasmota turned off.. so im not sure what it could be anymore <_>
> 
> --edit--
> I turned off every plugin except UI Customizer and the main tabs are still gone.. what do I do now??

Try clean the data of the UI Customizer and the set it up again - it might have broken something.
To cleanup find the UI Customizer in the Plugin manager and press the eraser icon: https://fontawesome.com/icons/eraser?style=solid

 Is it still a problem after cleanup?

LMS0815, 2020-12-02:
Looks like: https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/8

LazeMSS, 2020-12-03:
I figured it out - it was because of the fac that other plugins saved data without showing the UI - now my code checks for this. This has been fixed in the 0.1.0.0 release: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.0.0

closed 2020-12-03 by LazeMSS