LazeMSS/OctoPrint-UICustomizer #10: Option to have navbar to use the full webpage width

NovaViper, 2020-12-02:
It's a bit odd that the rest of the page expands but the navigation bar doesn't (Shown below)
[image]

The navbar should also use the full width of the browser just like the rest of the UI does

LazeMSS, 2020-12-02:
Strange, what browser and os?
What settings do you have turned on, fixed header?

Guilouz, 2020-12-02:
Same here, I use Chrome on MacOS. Fixed Header or not same issue.

When installing OctoPrint-WideScreen nav bar is ok

NovaViper, 2020-12-03:
> Strange, what browser and os?
> What settings do you have turned on, fixed header?

The browser is Vivaldi, and the OS is EndeavourOS KDE. I have all of the default settings on

LazeMSS, 2020-12-03:
Does this page behave the same:
https://getbootstrap.com/2.3.2/examples/fluid.html or is that full width?

NovaViper, 2020-12-03:
> Does this page behave the same:
> https://getbootstrap.com/2.3.2/examples/fluid.html or is that full width?

It's full width there

LazeMSS, 2020-12-03:
> > Does this page behave the same:
> > https://getbootstrap.com/2.3.2/examples/fluid.html or is that full width?
> 
> It's full width there

Try running the following in a developer console in your browser:
$('#navbar div.navbar-inner a.btn-navbar').remove();

Then run this:
$('#navbar > div.navbar-inner.default > div:first').prepend('<a class="btn btn-navbar collapsed" data-toggle="collapse" data-target=".UICMainMenu"><span class="icon-bar"></span><span class="icon-bar"></span><span class="icon-bar"></span></a>');

NovaViper, 2020-12-04:
Um.. how do I run it in the developer console? It's the same developer console as for Chrome (since the browser is based on Chrome)

LazeMSS, 2020-12-05:
> Um.. how do I run it in the developer console? It's the same developer console as for Chrome (since the browser is based on Chrome)

https://help.vivaldi.com/article/developer-tools/

LazeMSS, 2020-12-06:
Should be now fixed - please update and test :)

Guilouz, 2020-12-06:
No not working, same issue with latest update.

[image]

LazeMSS, 2020-12-06:
I'm sorry to say but there must be a plugin installed that messes with your installation.
This is how it looks in Vilvadi for me:
[image]

NovaViper, 2020-12-06:
It's not working for me either, even when I only have UI Customizer loaded

Guilouz, 2020-12-06:
> I'm sorry to say but there must be a plugin installed that messes with your installation.
> This is how it looks in Vilvadi for me:
> [image]

I use Google Chrome and it's working when I use Octoprint widescreen plugin so I think there is a mistake in your code.

Without Widescreen plugin, only UICustomizer : 

[image]

With Widescreen plugin + UICustomizer : 

[image]

LazeMSS, 2020-12-06:
I will look ASAP

LazeMSS, 2020-12-07:
I have looked and looked - it must be related in one of the plugins. The way the Widescreen plugin does the widescreen is not pretty. I would rather find the real source to the problem and fix it together with the author of that plugin. I know it sounds weird but that way my hack wont later on destroy somebody elses plugins - trying to keep it clean.

LazeMSS, 2020-12-07:
> It's not working for me either, even when I only have UI Customizer loaded

So you dont have widescreen - even when you just have UI Customizer loaded as 3rd party plugin? In which browsers?

knopancakes, 2020-12-07:
FYI, I am seeing the same behavior in Firefox 83.0 with OctoPrinti 1.5.1 and OctoPrint-UICustomizer 0.1.0.5

LazeMSS, 2020-12-07:
> FYI, I am seeing the same behavior in Firefox 83.0 with OctoPrinti 1.5.1 and OctoPrint-UICustomizer 0.1.0.5

Could you post a list of plugins you are using?

NovaViper, 2020-12-07:
> > It's not working for me either, even when I only have UI Customizer loaded
> 
> So it does not work in full widescreen? - even when you just have UI Customizer loaded as 3rd party plugin? In which browsers? Have you tried Chrome.

It didn't work in either maximixed mode or full screen, tried it on multiple computers (one with Chrome and one with Vivaldi), all have the same issue.

LazeMSS, 2020-12-07:
> It didn't work in either maximixed mode or full screen, tried it on multiple computers (one with Chrome and one with Vivaldi), all have the same issue.

Can you post a list of plugins?

knopancakes, 2020-12-07:
Full list of plugins running on my setup:
DisplayLayerProgress
Octolapse
PrintTimeGenius
Printer Dialogs
Printer Notifications
UI Customizer
Virtual Printer

LazeMSS, 2020-12-07:
> Printer Dialogs
> Printer Notifications

Cant find these plugins

NovaViper, 2020-12-07:
> Can you post a list of plugins?

I ran it after uninstalling all of the plugins

LazeMSS, 2020-12-07:
> I ran it after uninstalling all of the plugins

So you have no other than my plugin installed as a 3rd party plugin?

knopancakes, 2020-12-07:
> 
> 
> > Printer Dialogs
> > Printer Notifications
> 
> Cant find these plugins

Those are built in, not 3rd party

NovaViper, 2020-12-07:
> So you have no other than my plugin installed as a 3rd party plugin?

Yes

LazeMSS, 2020-12-07:
> > 
> > So you have no other than my plugin installed as a 3rd party plugin?
> 
> Yes

And still it does not work - I cant understand it - if anyone of you guys are willing 5 minutes with teamviewer would help alot

knopancakes, 2020-12-07:
I lied to you all about the plugins I had installed. The original comment has been edited to reflect only 3rd party plugins, including the Fullscreen Plugin I had missed.

LazeMSS, 2020-12-07:
> Fullscreen Plugin

Does it work uninstalling it - is it the fullscreen webcam?

knopancakes, 2020-12-07:
Yes, the Fullscreen Plugin is the webcam one. Disabling the Fullscreen Plugin resolves the oddities with the webcam widget, but not the top bar
[image]

LazeMSS, 2020-12-07:
> Yes, the Fullscreen Plugin is the webcam one. Disabling the Fullscreen Plugin resolves the oddities with the webcam widget, but not the top bar

I will try your list now

knopancakes, 2020-12-07:
I'll have to work on getting teamviewer installed. Got some other things I need to take care of first. Among them, hungry cats to feed.

Re: python, that also jumped out at me... @NovaViper is also running 2.7, but @Guilouz has 3.7 and also experiences the issue.

Settings:
[image]

[image]

[image]

LazeMSS, 2020-12-07:
Thanks - if you have the time try running this in your browsers console:
`  $('#navbar > div.navbar-inner > div:first').removeClass("container").addClass("container-fluid").removeAttr("style","");
   $('div.UICMainCont').removeClass("container").addClass("container-fluid");
    $('div.UICMainCont > div.row:first').removeClass("row").addClass("row-fluid");`

Guilouz, 2020-12-07:
> $('#navbar > div.navbar-inner > div:first').removeClass("container").addClass("container-fluid").removeAttr("style",""); $('div.UICMainCont').removeClass("container").addClass("container-fluid"); $('div.UICMainCont > div.row:first').removeClass("row").addClass("row-fluid");

With this in browser console it's working.

LazeMSS, 2020-12-07:
> With this in browser console it's working.

Thanks - I knew somewhere something was tampering with the layout - this is included in the next version then!

knopancakes, 2020-12-08:
Works for me too, thanks!

LazeMSS, 2020-12-08:
Thanks everyone for testing.

I will try and get the new release out soon.

LazeMSS, 2020-12-08:
Please test to confirm its fixed in the latest version 0.1.0.9

Guilouz, 2020-12-08:
> Please test to confirm its fixed in the latest version 0.1.0.9

Working with latest release. Thanks.

NovaViper, 2020-12-09:
It's works in the latest version for me as well! Thanks :D

closed 2020-12-09 by NovaViper