LazeMSS/OctoPrint-UICustomizer #9: Unwanted relod of CamWidget if Active Filters Extended (0.1.0) is installed

LMS0815, 2020-12-02:
If you have the _[Active Filters Extended (0.1.0)](https://github.com/jneilliii/OctoPrint-ActiveFiltersExtended)_ plugin installed, 
on every filter change at the terminal window the webcam widget is reset/reloaded .

LazeMSS, 2020-12-02:
HLS or normal webcam?

LMS0815, 2020-12-02:
If normal means PiCam, no.

Stream URL: `http://192.168.1.250:8080/mjpg/I4/video.mjpg?fps=10&q=50&cache=1&decode=1`

Disabling the plugin helps. 
I assume it has something to do with the plugin saving/changing OctoPrint configuration `config.yaml`.

LazeMSS, 2020-12-02:
yes correct - will look now 👍

LMS0815, 2020-12-02:
Add:

Occasionally the whole UI is corrupted (disappears) if the _Active Filters Extended_ is enabled.
Means only the header line is shown. As soon your update is available, I'll give it a test regarding this finding.

commit 98f9bc5e (2020-12-02): Fixed updates on settings saved without showing settings dialog. Close #9
closed 2020-12-02 by LazeMSS