LazeMSS/OctoPrint-UICustomizer #11: Main tabs not going full width

LordCorellon, 2020-12-04:
I recently installed the UI customizer and sadly it is unusable all the other main tabs  do not stretch full width.

The most noticeable is the terminal tab which will only take up a 1/3rd of the middle panel.   Panel is set up for width 6 but the terminal output is only as wide as the left panel (currently set for width two)

LazeMSS, 2020-12-04:
Could you provide more information, browser, os and maybe a screenshot

LordCorellon, 2020-12-04:
[image]
Windows 10 - Ver 2004 - Build 19401.572
Chrome Version 87.0.4280.88 (Official Build) (64-bit)

LordCorellon, 2020-12-04:
Plugin List:
Active Filters Extended (0.1.0)
Arc-Welder (1.0.0+u.bb71e8f)
Bed Visualizer (1.0.0)
BLTouch Plugin (0.3.4)
Change Filament Plugin (0.3.1)
Detailed Progress (0.2.6)
Enclosure Plugin (4.13.1)
ipOnConnect (0.2.4)
Marlin EEPROM Editor (2.0.0)
Multi Line Terminal (0.1.5)
PrintTimeGenius Plugin (2.2.6)
ProgressBasedOntime Plugin (0.1.6)
Simple Emergency Stop (1.0.4)
UI Customizer (0.1.0.0)

I did have the widescreen plugin and consolidated temp and control installed prior to UI Customizer but had uninstalled them

LazeMSS, 2020-12-05:
Thats strange im running an almost identical setup - windows 10 and chrome 87. I cant recreate the problems your having - this is how it looks on my pc:
[image]

Are you sure you have the rows/colums setup to give a total of 12?
Also maybe try disabling: "Multi Line Terminal" plugin.

Widescreen plugin should NOT be installed together with UI Customizer. Maybe I should add a warning about it.

Im looking into the  consolidated temp and control plugins - they do manipulate alot with the layout and more similar to my plugin.

LordCorellon, 2020-12-05:
The total is 12 for sure, 3+6+3.

Multi-line terminal works fine, the issue is completely within the consolidated temp and control plug, when I completely uninstall it (and clear data) then the issue goes away, if it's just disabled or uninstalled (without clearing data) then every tab gets squashed.

LazeMSS, 2020-12-05:
Yes im looking into it now - there is a problem - im fixing it.

LazeMSS, 2020-12-05:
Try pressing F12 (developer console) in Chrome and paste the following into the console:
`  $('div.page-container').css({'min-width':''});
                $('div.footer').css({'padding-left':'','padding-right':''});
                $('div.UICMainCont > div:first').css({'margin-left':'','padding-right':''});
                $('div.UICMainCont').removeClass('row-fluid');`

LordCorellon, 2020-12-05:
When I run it, it seems to do the opposite, makes everything in the middle frame squish about 10% more to the left.

LazeMSS, 2020-12-05:
Can you post a screenshot of your settings for consolidated ?

It might be easier to implement it's functionality into my plugin instead of working around these problems 🙈

LordCorellon, 2020-12-05:
[image]

Tried with and without nav bar resize enabled

LazeMSS, 2020-12-05:
Ahh you don't have themify. I can see the control plugin have different behaviour for this. I will spend more time investigating tomorrow. 

Is the only purpose of the control plugin to join the temperature and controls onto one tab?

LordCorellon, 2020-12-06:
Pretty much what I use it for, I always hated having to switch tabs when changing filament or adjusting offsets.  In fact I would love if the terminal could be combined on the same screen as well and hoped a plugin might do so later.

I'm hoping one day Octoprint moves to a modular sort of layout (If your familiar with OpnSense I like the dashboard setup) where you can specify columns width (like your plugin) but drag and drop "Widgets" to each area to arrange in the order you'd like

LazeMSS, 2020-12-06:
Should be now fixed - please update and test :)

Guilouz, 2020-12-06:
No not working, same issue with latest update

[image]

LazeMSS, 2020-12-07:
@LordCorellon  can you confirm if its working?

Guilouz, 2020-12-07:
There is issue with full screen. Look search bar

With UICustomizer :

[image]

With Widescreen plugin all is ok : 

[image]

LazeMSS, 2020-12-07:
> There is issue with full screen. Look search bar
> 
> With UICustomizer :
> 
> [image]
> 
> With Widescreen plugin all is ok :
> 
> [image]

Yes because widescreen plugin forces that - I would like to avoid that to keep the UI compliant to normal way of doing things. But this bug report is about the main tabs not showing full width when using "consolidated temp" plugin

LordCorellon, 2020-12-07:
@LazeMSS I can confirm the latest release has fixed the compatibility issue with Consolidated Temp and Control.   Works both in vertical and horizontal arrangements now.

Even better with the wider main tab the control fits perfectly in  horizontal mode without having to scroll down.

Thank you very much

LazeMSS, 2020-12-07:
> @LazeMSS I can confirm the latest release has fixed the compatibility issue with Consolidated Temp and Control. Works both in vertical and horizontal arrangements now.
>
> Even better with the wider [...]

Thanks - I have made a couple of improvements in the next aswell.

closed 2020-12-07 by LazeMSS

Guilouz, 2020-12-08:
> > There is issue with full screen. Look search bar
> 
> Yes because widescreen plugin forces that - I would like to avoid that to keep the UI compliant to normal way of doing things. But this bug report is about the main tabs not showing full width when using "consolidated temp" plugin.
> 
> Edit:
> Ahh noticed the search bar now - sorry but that when you have disabled "responsive/mobile" mode - I would not recommend turning that off - but I have a fix ready

Yes but I prefer the old settings panel on the left, that's why I have disable it.

LazeMSS, 2020-12-08:
> > > There is issue with full screen. Look search bar
> > 
> > 
> > Yes because widescreen plugin forces that - I would like to avoid that to keep the UI compliant to normal way of doing things. But this bug report is about the main tabs not showing full width when using "consolidated temp" plugin.
> > Edit:
> > Ahh noticed the search bar now - sorry but that when you have disabled "responsive/mobile" mode - I would not recommend turning that off - but I have a fix ready
> 
> Yes but I prefer the old settings panel on the left, that's why I have disable it. Maybe an option to choose it or not ?

Only way to have a settings menu that works on small screens :(