LazeMSS/OctoPrint-UICustomizer #13: UI issues with Other Plugins

holocronology, 2020-12-04:
I've been having multiple UI issues since the release of UI Customizer.

I initially reported [here ](https://github.com/jneilliii/OctoPrint-ConsolidateTempControl/issues/21)and [here](https://github.com/OctoPrint/OctoPrint/issues/3850).

Specifically, there seems to be some incompatibility going on with Consolidated Temp Control and Consolidated Tabs. The latter seems to be working now for whatever reason. But the former will not even install. It shows as installing successfully but never appears in the plugins. If UI Customizer is uninstalled, everything functions normally.

LazeMSS, 2020-12-04:
Have you tried deleting the data from my plugin
But in general there might be conflicts because the plugins tries to manipulate with the same content.

I will try and install consolidated tabs later and test it.

LazeMSS, 2020-12-05:
Consolidated Tabs is working I can confirm that. And I have a fix on the way for Consolidated Temp Control

LazeMSS, 2020-12-06:
Should be now fixed - please update and test :)

holocronology, 2020-12-06:
Thanks. I'll update and test today. 

Is it possible to add a feature--I guess I should post this elsewhere--to the Consolidated Temp/Control that turns off the webcam view in the center area if it is added to a sidebar?

holocronology, 2020-12-06:
Still being weird.

https://i.imgur.com/RVNv1ck.png

LazeMSS, 2020-12-06:
Are your running themify plugin ?

holocronology, 2020-12-09:
I ended up doing a fresh install of Octo.

Consolidated Temp Control still exhibits the same cramming when set to "horizontal" layout. It's working correctly in vertical.

I ended up switching to Consolidated Tabs, since it afforded the same functionality but better control and options. This seems to be working OK.

LazeMSS, 2020-12-09:
Okay closing

closed 2020-12-09 by LazeMSS