LazeMSS/OctoPrint-UICustomizer #15: Plugin icon for the OctoPrint-WifiStatus Plugin misaligned?

devsfan1830, 2020-12-04:
Seems to be similar to an older solved bug. The icon for this plugin is centered rather than being aligned with the other icons. No biggie but if you could push a fix whenever, or let me know how to do so myself I'd appreciate it. Otherwise just want to say this is a fantastic plugin. Gives the web interface a much needed refresh.
[image]

LazeMSS, 2020-12-05:
This is probally fixed in the next release - you can try and pull the latest commit if you dont want to wait for the release. I dont have wifi on my RPI so dont know how it behaves when connected. I tried faking it a bit and seems to be working. When tested please close this.

LazeMSS, 2020-12-06:
Should be now fixed - please update and test :)

devsfan1830, 2020-12-06:
Unfortunately the update didnt fix it. Tried flushing my browser history. Disabling and re-enabling both plugins. Its really no biggie. Its not a vital plugin anyways. I just used it to double check my connection once and just left it enabled. No other plugin i use place icons in that bar so I was just letting you know in case others might. Im happy to close this issue for now if you want.

LazeMSS, 2020-12-07:
@devsfan1830  can you try and disabling all other 3rdparty plugins you have installed:
This is a list of plugins I can confirm it works with on my setups
- Autoscroll
- Autoselect Plugin
- Filament Sensor Simplified
- Fix CBD Firmware Plugin
- Navbar Temperature Plugin
- PortLister
- Printoid Plugin
- PrintTimeGenius Plugin
- Themeify
- UI Customizer

devsfan1830, 2020-12-07:
My currently installed plugins:
- DisplayLayerProgress Plugin
- OctoPrint-WifiStatus
- Printoid Plugin 
- PrintTimeGenius Plugin
- PrusaSlicer Thumbnails
- Themify
- UI Customizer

I have disabled everything. Then reenabled UI Customizer only, reloaded server. Then enabled WifiStatus. WifiStatus icon still ends up dead center.

LazeMSS, 2020-12-07:
I have just uninstalled everything aswell - this is my list of 3rd party:
- OctoPrint-WiFiStatus
 -UI Customizer

And built-in/bundled:
- Action Command Notification Support
 -Action Command Prompt Support
 -Announcement Plugin
 -Anonymous Usage Tracking
 -Application Keys Plugin
 -Backup & Restore
 -Core Wizard
 -Discovery
 -Error Tracking
 -File Check
 -Firmware Check
 -GCode Viewer
 -Logging
 -Pi Support Plugin
 -Software Update
 -Virtual Printer

devsfan1830, 2020-12-07:
Might figured it out. Oversight on my part. I didnt see the "Center Top Bar Icons"
 option. Turned that off and refreshed and its all good now. Sorry! I did however observe an odd behavior with the "Preview" option turned on. Every time i toggled the center option it dumped the contents of that wifi status drop down into the top bar. Fortunately a refresh clears its away so its not a permanent thing. Here a video in case you're curious.  https://www.youtube.com/watch?v=FldWvQk8Q8U But as far as this issue is concerned I'll go ahead and close it. Thanks!

closed 2020-12-07 by devsfan1830

LazeMSS, 2020-12-07:
> Might figured it out. Oversight on my part. I didnt see the "Center Top Bar Icons"
> option. Turned that off and refreshed and its all good now. Sorry! I did however observe an odd behavior with the [...]

DOHHH

And I found the other you found and fixed that in the next release