LazeMSS/OctoPrint-UICustomizer #16: Dropdown Menu Disappears Off Screen

ruralaccent, 2020-12-06:
Hi. Great plugin. I'm already using it on two instances, but I have an issue where the dropdown plugins menu in the settings window disappears offscreen and I have to reduce browser scale size to be able to access all of the drop down menu items. 

I run 4k monitors on all of my computers. Octoprint 1.5.1.

Can a scroll bar be added to the menu or similar, or another fix? Or perhaps just provide an option to turn off the condensed settings page and let the original 'stock' one show instead? 

With scale at 100%:

[image]

When scaled down:

[image]

LazeMSS, 2020-12-06:
have you tried the latest version? 0.1.0.5 - it should help with these issues

LazeMSS, 2020-12-06:
Confirmed - it is a bug :)

ruralaccent, 2020-12-06:
I was just about to say, it's 0.1.0.5 that I'm running. I know I can deselect the mobile responsive box, but I do prefer the new settings layout so would like to use it, if possible. Many thanks :)

LazeMSS, 2020-12-06:
I have it fixed in the next release - comming soon :)

ruralaccent, 2020-12-07:
Great. Thank you.

LazeMSS, 2020-12-08:
Please test to confirm its fixed in the latest version 0.1.0.9

LazeMSS, 2020-12-09:
@ruralaccent  fixed or not?

ruralaccent, 2020-12-09:
Sorry. I've only just had a chance to update the plugin this morning. I'll let you know ASAP.

ruralaccent, 2020-12-09:
The fix works on both of my OctoPrint systems and when view by three different computers. The only issue  is hat the final item in the dropdown, whilst visible and clickable, is partially cut off. 

[image]

It's not a big issue and doesn't affect usability.

LazeMSS, 2020-12-09:
Ahh I will fix that last part in next update. Will close for now.

closed 2020-12-09 by LazeMSS

ruralaccent, 2020-12-09:
Many thanks for your work on this :)