LazeMSS/OctoPrint-UICustomizer #17: The webcam widget never loads the video

hobbieman, 2020-12-07:
I have the webcam working.  I can see it just fine in the "Control" tab.  However, the webcam widget that UI Customizer adds to the main page just keeps saying "Loading Webcam..."

LazeMSS, 2020-12-07:
1. HLS or normal webcam?
2. Which plugins do you have enabled? Try disabling anyone related to webcams etc.

hobbieman, 2020-12-07:
I am using a the raspberry pi camera, and the software at http://elinux.org/RPi-Cam-Web-Interface

I have disabled all plugins that did not come standard with Octoprint, though all the plugins I had added did not have anything to do with the camera, but I disabled them all anyway and I am still not able to see the camera in the UI customizer widget.

LazeMSS, 2020-12-07:
I don't know the difference in that - if you just enter the URL from that software into the webcam url in octoprint there should be no difference. Does anything happen when you switch to and from octoprint ui?

If running chrome or any other browser with a developer console (in Windows Chrome pressing F12 opens the developer console) try and copy paste the ouput from the console.

hobbieman, 2020-12-08:
> Does anything happen when you switch to and from octoprint ui?

Sorry, not sure what you mean there. could you explain please?

> And just out of curosity why dont you use the native rpi camera option in octoprint?

I played around with the Pi and the camera for some time before I discovered OctoPrint. So I just continued to use the set up with OctoPrint.  I was not aware that Octoprint could access the camera "natively".  How do I do that?

> If running chrome or any other browser with a developer console (in Windows Chrome pressing F12 opens the developer console) try and copy paste the ouput from the console.

Here is the output I get:

packed_libs.js?df208d52:1219 Less has finished and no sheets were loaded.
[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
[DOM] Input elements should have autocomplete attributes <attribute>: (More info: <URL>) 
packed_core.js?3fb354c1:850 Starting dependency resolution...
packed_core.js?3fb354c1:860 ... dependency resolution done
packed_core.js?3fb354c1:883 Initial application setup done, connecting to server...
packed_core.js?3fb354c1:702 Connected to the server
packed_core.js?3fb354c1:707 Triggering connect on all view models
packed_core.js?3fb354c1:709 Server (re)connect processed
packed_core.js?3fb354c1:883 Finalizing application startup
packed_core.js?3fb354c1:866 Going to bind 38 view models...
packed_core.js?3fb354c1:876 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?3fb354c1:876 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?3fb354c1:876 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?3fb354c1:876 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?3fb354c1:158 User pi logged in
packed_core.js?3fb354c1:833 Error calling onAllBound on view model UICustomizerViewModel : TypeError: Cannot read property 'pause' of undefined
    at UICustomizerViewModel.self.CustomW_initWebCam (http://192.168.123.173:5000/static/webassets/packed_plugins.js?eb337d99:538:44)
    at String.<anonymous> (http://192.168.123.173:5000/static/webassets/packed_plugins.js?eb337d99:237:68)
    at Function.each (http://192.168.123.173:5000/static/webassets/packed_libs.js?df208d52:1:2573)
    at w.fn.init.each (http://192.168.123.173:5000/static/webassets/packed_libs.js?df208d52:1:1240)
    at Array.<anonymous> (http://192.168.123.173:5000/static/webassets/packed_plugins.js?eb337d99:223:32)
    at Function.each (http://192.168.123.173:5000/static/webassets/packed_libs.js?df208d52:1:2573)
    at w.fn.init.each (http://192.168.123.173:5000/static/webassets/packed_libs.js?df208d52:1:1240)
    at UICustomizerViewModel.self.set_rowsLayout (http://192.168.123.173:5000/static/webassets/packed_plugins.js?eb337d99:204:29)
    at UICustomizerViewModel.self.UpdateLayout (http://192.168.123.173:5000/static/webassets/packed_plugins.js?eb337d99:146:26)
    at UICustomizerViewModel.self.onAllBound (http://192.168.123.173:5000/static/webassets/packed_plugins.js?eb337d99:76:26)
callViewModelIf @ packed_core.js?3fb354c1:833
(anonymous) @ packed_core.js?3fb354c1:826
Pn @ packed_libs.js?df208d52:1
(anonymous) @ packed_libs.js?df208d52:1
callViewModelsIf @ packed_core.js?3fb354c1:826
callViewModels @ packed_core.js?3fb354c1:825
bindViewModels @ packed_core.js?3fb354c1:881
setTimeout (async)
(anonymous) @ packed_core.js?3fb354c1:883
u @ packed_libs.js?df208d52:1
fireWith @ packed_libs.js?df208d52:1
k @ packed_libs.js?df208d52:1
(anonymous) @ packed_libs.js?df208d52:1
load (async)
send @ packed_libs.js?df208d52:1
ajax @ packed_libs.js?df208d52:1
OctoPrintClient.ajax @ packed_client.js?c542e09b:7
OctoPrintClient.get @ packed_client.js?c542e09b:7
OctoPrintSettingsClient.get @ packed_client.js?c542e09b:55
SettingsViewModel.self.requestData @ packed_core.js?3fb354c1:240
fetchSettings @ packed_core.js?3fb354c1:883
setTimeout (async)
(anonymous) @ packed_core.js?3fb354c1:883
u @ packed_libs.js?df208d52:1
fireWith @ packed_libs.js?df208d52:1
k @ packed_libs.js?df208d52:1
[... 11 lines cut ...]
(anonymous) @ packed_core.js?3fb354c1:883
u @ packed_libs.js?df208d52:1
fireWith @ packed_libs.js?df208d52:1
o.<computed> @ packed_libs.js?df208d52:1
DataUpdater.self._onConnectMessage @ packed_core.js?3fb354c1:702
(anonymous) @ packed_client.js?c542e09b:13
Pn @ packed_libs.js?df208d52:1
(anonymous) @ packed_libs.js?df208d52:1
OctoPrintSocketClient.propagateMessage @ packed_client.js?c542e09b:13
(anonymous) @ packed_client.js?c542e09b:20
(anonymous) @ packed_libs.js?df208d52:1
_t @ packed_libs.js?df208d52:1
(anonymous) @ packed_libs.js?df208d52:1
(anonymous) @ packed_libs.js?df208d52:1
onMessage @ packed_client.js?c542e09b:20
r.dispatchEvent @ packed_libs.js?df208d52:1206
(anonymous) @ packed_libs.js?df208d52:1206
r._transportMessage @ packed_libs.js?df208d52:1206
r.emit @ packed_libs.js?df208d52:1206
ws.onmessage @ packed_libs.js?df208d52:1206
packed_core.js?3fb354c1:881 ... binding done
packed_core.js?3fb354c1:881 Application startup complete
packed_core.js?3fb354c1:866 [Violation] 'setTimeout' handler took 734ms
[Violation] Forced reflow while executing JavaScript took 41ms
packed_libs.js?df208d52:1 [Violation] 'load' handler took 414ms

hobbieman, 2020-12-08:
Well, I upgraded to the newest OctoPrint (v1.5.1) and the webcam widget started working on its own.  Thanks.

LazeMSS, 2020-12-08:
> Well, I upgraded to the newest OctoPrint (v1.5.1) and the webcam widget started working on its own. Thanks.

I'm closing the issue then

closed 2020-12-08 by LazeMSS