LazeMSS/OctoPrint-UICustomizer #18: cannot access FullScreen webcam plugin with "Hide Main Camera" checked/buggy pop-out action if unchecked

jusbot, 2020-12-07:
https://github.com/BillyBlaze/OctoPrint-FullScreen

Plugin's usual behavior is to launch a full screen webcam feed when double-clicking on the webcam window under the control tab. With "hide main camera" checked off, there is no way to access - double-clicking on the widget feed highlights it, and when popped out it will not go to full screen. I tried turning the main camera back on and I can launch full screen fine, but I get an extra pop-out button and the video does not pop back in properly after without refreshing the interface.

knopancakes, 2020-12-07:
This may be related. I find even opening the menu has scaling issues:
[image]

knopancakes, 2020-12-07:
There's pretty inconsistent behavior with the full screen webcam. Sometime it opens correctly, although outside of screen borders, and needs to be dragged / resized to find the minimize button. Other times it is sized like the OctoPrint Setting dialogue in the screenshot above, but contains no video. I tested this with and without "hide main camera" checked and experience the issue with both configurations. Refreshing the page seems randomly determine the behavior.
[image]

I also notice that the Tool and Bed temperature under the Webcam feed do not track the actual temperatures when I'm running a print. Will try to get a screenshot of that next time I see it. Seems like checking "hide main camera" causes these fields to be hidden as well.

knopancakes, 2020-12-07:
Full list of plugins running on my setup:
DisplayLayerProgress
Octolapse
PrintTimeGenius
Printer Dialogs
Printer Notifications
UI Customizer
Virtual Printer

LazeMSS, 2020-12-07:
These bug repports are not related but I will look into it

LazeMSS, 2020-12-07:
> https://github.com/BillyBlaze/OctoPrint-FullScreen
>
> Plugin's usual behavior is to launch a full screen webcam feed when double-clicking on the webcam window under the control tab. With "hide main [...]

I will look into your bug report later this week

knopancakes, 2020-12-07:
Sorry to waste everyone's time, I did not realize I also had Fullscreen Plugin (0.0.6) installed. The behavior I reported here goes away after disabling the Fullscreen Plugin.

LazeMSS, 2020-12-13:
@jusbot

LazeMSS, 2020-12-14:
Closing - confirmed fixed by others. @jusbot  feel free to reopen if you disagree :)

closed 2020-12-14 by LazeMSS

jusbot, 2020-12-14:
> Closing - confirmed fixed by others. @jusbot feel free to reopen if you disagree :)

Yup, resolved, works great from the main cam. However, it does not work from your widget, making it so I have to keep the redundant main webcam view on to use it. Maybe for a future update it could either work with your widget or you could incorporate a similar function? Thanks!

(My use case is that I run a makerspace in a middle school, so I'm trying to streamline octoprint as much as possible to make it easier to understand for students. I have a big TV in the main part of the makerspace that I sometimes throw the full-screen cam up on to draw folks in.)

LazeMSS, 2020-12-15:
> Yup, resolved, works great from the main cam. However, it does not work from your widget, making it so I have to keep the redundant main webcam view on to use it. Maybe for a future update it could [...]

What does not work - the fullscreen plugin or?

jusbot, 2020-12-15:
Sorry, it might make more sense with screenshots.
Here's my setup... I have the "Hide main camera" box unchecked, but would like to check it off to get rid of the redundant view.
[image]

With it on, I can double-click on the main camera which expands to this:
[image]

Then I can double-click it again to access the full screen plugin's view in the window:
[image]

then I can click on the expand arrows in the lower right and it gets me the full screen plugin actually in full screen (looks great on the TV).

If I double-click on the UI Customizer camera view, it does not open the full screen plugin's view with the data overlay. It also does not expand to full screen. 

So UI Customizer is no longer breaking the plugin (yay!) but it is not providing access to it from your camera view. If you added a full screen view button on your webcam pop-out it would make the full screen plugin irrelevant.

Does that make more sense? Thanks!

LazeMSS, 2020-12-15:
Thanks - I will look into fixing the fullscreen stuff :)