LazeMSS/OctoPrint-UICustomizer #19: Main tab setting issue

Guilouz, 2020-12-08:
The new setting for customize main tab doesn't show all tabs, it's only show default tab, not others from plugins

With UICustomizer :

[image]

With Tab Order plugin : 

[image]

knopancakes, 2020-12-08:
Seeing this affect the Octolapse plugin as well.
Octolapse used to be the left-most tab, now it is the right-most. There does not appear to be a way to move it or change the icon like the other main tabs.
[image]

LazeMSS, 2020-12-09:
Dohhh - I will fix it as soon as possible.

LazeMSS, 2020-12-09:
Please test if fixed in latest release :)

@Guilouz i also added color picker and some more stuff

Guilouz, 2020-12-09:
> Please test if fixed in latest release :)
> 
> @Guilouz i also added color picker and some more stuff

All plugins tabs now appears but color picker doesn't work, it can be changed but there no effect on the main tab.

LazeMSS, 2020-12-10:
Ahh - btw. should the labels also be colored or only the icons?

Guilouz, 2020-12-10:
> Ahh - btw. should the labels also be colored or only the icons?

Only icons.

LazeMSS, 2020-12-10:
Please test with latest

Guilouz, 2020-12-10:
> Please test with latest - is it fixed?

yes it's working now.

closed 2020-12-10 by LazeMSS