LazeMSS/OctoPrint-UICustomizer #20: No settings wrench on iphone

simpat1zq, 2020-12-09:
[
 ENTER DESCRIPTION HERE- ALSO ADD SCREENSHOT IF POSSIBLE!
 Describe your problem?
 What is the problem?
 Can you recreate it?
 Did you try disabling plugins?
 Did you remeber to update the subject?
]

**Plugins installed:**
- DeleteAfterPrint
- PrintJobHistory
- PrintTimeGenius
- UltimakerFormatPackage
- action_command_notification
- action_command_prompt
- announcements
- astroprint
- backup
- bedlevelingwizard
- bedlevelvisualizer
- cancelobject
- errortracking
- gcodeviewer
- klipper
- mqtt
- navbartemp
- octopod
- pluginmanager
- preheat
- resource_monitor
- softwareupdate
- statefulsidebar
- tplinksmartplug
- uicustomizer
- virtual_printer

**Software versions:**
- OctoPrint 1.5.1
- Python 3.6.9

LazeMSS, 2020-12-09:
Could you try and describe a bit more?
What happens if you swipe to the right - is the wrench icon anywhere to be seen?

simpat1zq, 2020-12-09:
I’ve tried all sorts of tapping everywhere, swiping, and rotating the screen. I can’t get it to show.

simpat1zq, 2020-12-10:
I updated to the latest version 0.1.1.1 and I still don't see the wrench. The top bar does look cleaner as there is nothing overflowing below it. It also doesn't show the temps when I have the phone in portrait mode, but the show up in landscape. The wrench just doesn't show up.

LazeMSS, 2020-12-10:
> I updated to the latest version 0.1.1.1 and I still don't see the wrench. The top bar does look cleaner as there is nothing overflowing below it. It also doesn't show the temps when I have the phone in portrait mode, but the show up in landscape. The wrench just doesn't show up.
One or more of the plugins you have installed add a lot of stuff to the menu or topbar.

LazeMSS, 2020-12-13:
@simpat1zq  please test with latest version (https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.2)

simpat1zq, 2020-12-13:
Still doesn’t show up. And now if ‘compact navbar’ is enabled it doesn’t show the temperature number, only the icons. If I turn that off it shows the temperature. But still no settings wrench.

LazeMSS, 2020-12-13:
Here is how it looks on my iphone - could you look into your plugins and try disabling them - as I wrote earlier:
"One or more of the plugins you have installed add a lot of stuff to the menu or topbar. I can try and install them all but if you have a guess which one it would be easier"

[image]

LazeMSS, 2020-12-14:
@simpat1zq  What menu settings do you have enabled in UI Customizer... Im guessing you dont have enabled the responsive option?

simpat1zq, 2020-12-14:
I havent had a chance to disable plugins yet. Im printing an order and its going to take another day. The only thing i have unchecked is fixed header and fixed footer.

LazeMSS, 2020-12-15:
@simpat1zq  should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.5 - if not then reopen please 👍

simpat1zq, 2020-12-15:
I updated. It didn't work until I disabled 'Center Topbar Icons' and 'Improve Mobile/Responsiveness'. And then I see the wrench. Enabling Compact Menu doesn't seem to do anything. It still shows the wrench/power/username icons separately. The compact menu doesn't seem to work on mobile or on desktop. Let me know if you want me to create a new issue for this or if you consider it the same issue.

LazeMSS, 2020-12-16:
And you are running on the latest version? I would not recommend running without improve mobile if you use it from a mobile.

The wrench and other menu items are always on the right hand side. The compact menu only works when in mobile responsive mode.

Try pressing the report a bug in settings and I will get all the data and I can test but I think my plugin is working as I expect but not like you want it to 😄

Maybe drop a screenshot etc

simpat1zq, 2020-12-16:
This is on the latest version of this plugin as of 10 minutes ago. I disabled most plugins. There are no icons now other than the octoprint logo on the top left. 
[image]

I had to pull the bug report from my PC. I get the same issue in safari iOS and the chrome app on iOS, which is probably expected. 

Please let me know if there is any other information you might need. 
[
 ENTER DESCRIPTION HERE- ALSO ADD SCREENSHOT IF POSSIBLE!
 Describe your problem?
 What is the problem?
 Can you recreate it?
 Did you try disabling plugins?
 Did you remeber to update the subject?
]

**Plugins installed:**
- action_command_notification
- action_command_prompt
- announcements
- backup
- gcodeviewer
- mqtt
- pluginmanager
- softwareupdate
- uicustomizer

**Plugin settings:**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- disableTermInactive: true
- fixedFooter: false
- fixedHeader: false
- fluidLayout: true
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions:**
- OctoPrint 1.5.2
- Python 3.6.9

**Browser:**
-Mozilla/5.0 (Windows NT 10.0 Win64 x64 rv:83.0) Gecko/20100101 Firefox/83.0

LazeMSS, 2020-12-16:
Please update to the latest version: 0.1.1.5 - please try and update.

Goto Settings -> Software update: Advanced options - and click the force check

simpat1zq, 2020-12-16:
I’m already on that version

LazeMSS, 2020-12-16:
I have the identical settings on my iPhone and I cant recreate that no matter how hard I try :(

simpat1zq, 2020-12-16:
There might be something strange going on with my OP install then. It must have started recently too. The files take like 20 seconds to load. I disabled your plugin and it is still the same way.

LazeMSS, 2020-12-16:
Take a look in the developer console with your pc or the logs in the settings

simpat1zq, 2020-12-17:
So I figured out that there was a plugin from a while ago that added something in the announcements section. And the place they were pulling from is down so it didn’t load some stuff until that time out. 

After disabling the announcement, octoprint is back to normal, but I still don’t see any icons at the top with this plugin. I even removed the plugin with a clean uninstall and reinstalled it and the same thing happens.

LazeMSS, 2020-12-17:
> So I figured out that there was a plugin from a while ago that added something in the announcements section. And the place they were pulling from is down so it didn’t load some stuff until that time [...]

And nothing in the developer console or anything else - does it look okay if you resize your browser size on a computer?

simpat1zq, 2020-12-17:
Here’s some output from the dev console. If I make my desktop browser smaller, the icons for octoprint disappear. And the dev console doesn’t say anything new when I resize. 

Firefox can’t establish a connection to the server at ws://192.168.45.104:5000/sockjs/451/5eig4l2r/websocket. packed_libs.js:1232:25077
Some cookies are misusing the recommended “SameSite“ attribute 59
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.
3 192.168.45.104:5000
Password fields present on an insecure (http://) page. This is a security risk that allows user login credentials to be stolen.
192.168.45.104:5000
This page uses the non standard property “zoom”. Consider using calc() in the relevant property values, or using “transform” along with “transform-origin: 0 0”. 192.168.45.104:5000
Less has finished and no sheets were loaded. packed_libs.js:1249:143043
HTTP “Content-Type” of “text/html” is not supported. Load of media resource http://192.168.45.104:5000/ failed.

LazeMSS, 2020-12-17:
Do you have an external octoprint I can try on?

simpat1zq, 2020-12-17:
No not external. Did you want to do a screen share?

LazeMSS, 2020-12-17:
> No not external. Did you want to do a screen share?

Yes if possible email me at:
uicust
at
jemi
.
dk

remove newlines and replace at with @ 👍

LazeMSS, 2020-12-20:
Bug found - next release will fix this :)

LazeMSS, 2020-12-21:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.7
Please try and if not fixed then reopen this issue.

closed 2020-12-21 by LazeMSS