LazeMSS/OctoPrint-UICustomizer #22: Tabs not showing on one install, but do on another

ruralaccent, 2020-12-09:
I have two seperate Pi4s running the same version of Octoprint, all of the same plugins set up in the same way. They both have the same version of UI Customiser (latest version as of today's date, but while on one install UI Customiser displays all plugin tabs in the 'main tab' menu, the other install only shows the core tabs. I've scoured all of my settings and can't see anything obvious that's different and that would account for this...

Displays ok on this one:

[image]
 
Doesn't display on this one: 

[image]

**Plugins installed:**
- BetterHeaterTimeout
- DisplayLayerProgress
- GcodeEditor
- IFTTT
- PrintTimeGenius
- PrinterAlerts
- PrusaLevelingGuide
- action_command_notification
- action_command_prompt
- announcements
- backup
- backupscheduler
- bedlevelvisualizer
- cancelobject
- costestimation
- dashboard
- discovery
- enclosure
- errortracking
- excluderegion
- filamentmanager
- floatingnavbar
- gcodeviewer
- navbartemp
- pi_support
- pluginmanager
- preheat
- printoid
- prusaslicerthumbnails
- simpleemergencystop
- softwareupdate
- taborder
- uicustomizer
- virtual_printer

**Software versions:**
- OctoPrint 1.5.1
- Python 2.7.16
- Pi4

ruralaccent, 2020-12-09:
I toggled the tabs setting in UI Customiser and also updated the tabs in Tab Order plugin, and then disabled it, and now the correct tabs show in UI Customiser. I guess there must be a cache issue or something? I'm not sure if that's something that needs to be consider in revisions to UI Customier or an external issue. Feel free to close this if you think it's external.

LazeMSS, 2020-12-09:
yearh its a "bug" kinda - its fixed in next hotfi

LazeMSS, 2020-12-09:
Please test if fixed in latest release :)

ruralaccent, 2020-12-10:
So far as I can see, the issue isn't repeating.

closed 2020-12-10 by LazeMSS