LazeMSS/OctoPrint-UICustomizer #23: Drop down elements of header icons are hidden.

StaticNoise64, 2020-12-10:
New install of Octoprint

OctoPrint 1.5.2
Python 3.7.3
OctoPi 0.17.0

Additional Plugins: 
OctoPrint-WiFiStatus (1.4.0)
UI Customizer (0.1.1.1)

I found an error with the WiFi-Status plugin installed.

Using Chrome to inspect the elements I found the style of UIHeaderIcons is set to "Overflow: hidden" this is interfering with the dropdown menu of the plugin.

LazeMSS, 2020-12-10:
Thanks will fix ASAP

ertraid, 2020-12-10:
Hi. I think I stumbled against the same bug... just adding here in case it helps you.
In my case I am using MQTTPubilsh plugin (https://plugins.octoprint.org/plugins/mqttpublish/) which also displays a dropdown from the header bar, and they seem to be hidden when combined with UI customizer. StickyPad plugin seems to work fine though..

Great work with your plugin by the way!
ET

ertraid, 2020-12-10:
Also, I only noticed this issue in the last day or 2.. Cannot swear, but I think it was working fine before.
ET

ManuelMcLure, 2020-12-12:
I'm the author of OctoPrint-WiFiStatus and this was reported to me as well. I discovered that disabling the "Improve mobile/responsive" option resolves this issue, in case that helps debug this.

LazeMSS, 2020-12-13:
> I'm the author of OctoPrint-WiFiStatus and this was reported to me as well. I discovered that disabling the "Improve mobile/responsive" option resolves this issue, in case that helps debug this.

Its fixed in the release - btw. I can't fork your plugin but I made a bunch of changes to it to improve the UI - have a look at: https://github.com/LazeMSS/OctoPrint-WiFiStatus

LazeMSS, 2020-12-13:
Please test if the latest version(https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.2) does not fix it.

LazeMSS, 2020-12-13:
Reported as fixed in Wifi plugin issue - closing

closed 2020-12-13 by LazeMSS

StaticNoise64, 2020-12-13:
Sorry didn't get around to testing until today. That fixed the issue, thank you! Loving the plugin, good work!

ertraid, 2020-12-17:
Agreed… solved issue with MQTT Publish plugin as well. Thanks!