LazeMSS/OctoPrint-UICustomizer #24: Plugin is hanging if loading of webcam widget is delayed.

LMS0815, 2020-12-11:
My OrcoPrint webcam is configured loading an URL (`http://192.168.1.250:8080/mjpg/I4/video.mjpg?fps=10&q=50&cache=1&decode=1`) 

If the cam is not available the widget is displaying a spinner.
[image]

Unfortunately the rest of the plugin is not loaded/updated (e.g. main tab) until the spinner is gone.
As the cam is not available the spinner is shown forever and the whole plugin is stuck.
[image]

If I disable the cam widget, everything is perfect.
[image]

**Plugins installed:**
- action_command_notification
- action_command_prompt
- active_filters_extended
- announcements
- arc_welder
- backup
- bedlevelvisualizer
- costestimation
- discovery
- errortracking
- firmwareupdater
- gcodesystemcommands
- huecommands
- m73progress
- pluginmanager
- preheat
- resource_monitor
- softwareupdate
- tracking
- uicustomizer
- virtual_printer

**Software versions:**
- OctoPrint 1.5.2
- Python 2.7.14

LazeMSS, 2020-12-11:
Is the main cam disabled?

I have tried with a broken url and my keeps spinning and the rest of the UI loads fine. There shouldn't be anything blocking anything - its just set the url and that spinner is shown until an image/video is loaded

LMS0815, 2020-12-11:
Yes main cam screen is disabled.

LMS0815, 2020-12-11:
I found the rest of the plugin is not loading, as it was not showing the changes I made to the main tab.
Otherwise I would not have recognized it...
If possible try getting the spinner and make some changes to the main tabs (e.g. show text+icon). 
Reload again. The changes are not processed.

LazeMSS, 2020-12-11:
Works fine on my installation - can you try and explain step by step?

LazeMSS, 2020-12-11:
If you are running chrome can you try and look in the developers tools and console.

https://developers.google.com/web/tools/chrome-devtools#:~:text=Open%20DevTools,-There%20are%20many&text=Or%20press%20Command%20%2B%20Option%20%2B%20C,straight%20into%20the%20Console%20panel.

LazeMSS, 2020-12-13:
@LMS0815 I really cant recreate this - tried alot of things

LMS0815, 2020-12-14:
The issue seems to have something to do with my webcam (BlueIris stream). If I enter an invalid web cam address it's all working as expected. If I enter a valid cam address which is already blocked by an other browser window the widget is stuck. 
I believe this is an issue with the streaming app not sending an expected reply... ,
As this is not a common problem I would agree if you decide concentrating on issues shared by lots of users than on a this special setup.

Thank you for spending your time evaluating the issue.

LazeMSS, 2020-12-14:
No problem - and thanks for getting back 👍

closed 2020-12-14 by LazeMSS