LazeMSS/OctoPrint-UICustomizer #25: Webcam stuck on 'loading'

ruralaccent, 2020-12-11:
I've got two all-but identical OctoPrint systems, both running on Pi4s. On one, the webcam tab and side bar works fine. On the other, the webcam just hangs with a constant 'loading' sign. I also can't view the webcam on my Pi octodash screen. The cam worked fine on this system before I installed UI Customiser. I never noticed the issue before as the problem system is the one I used less often. Is is possible/likely that the UI plugin is the problem? 

**Plugins installed:**
- BetterHeaterTimeout
- DisplayLayerProgress
- GcodeEditor
- IFTTT
- PrintTimeGenius
- PrinterAlerts
- PrusaLevelingGuide
- action_command_notification
- action_command_prompt
- announcements
- backup
- backupscheduler
- bedlevelvisualizer
- cancelobject
- costestimation
- dashboard
- discovery
- enclosure
- errortracking
- excluderegion
- filamentmanager
- floatingnavbar
- gcodeviewer
- navbartemp
- pi_support
- pluginmanager
- preheat
- printoid
- prusaslicerthumbnails
- simpleemergencystop
- softwareupdate
- uicustomizer
- virtual_printer

**Software versions:**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

LazeMSS, 2020-12-11:
Any difference in the plugins installed on the two?

Different webcam streams? Can you try use the working installation webcam stream on the broken installation?

LazeMSS, 2020-12-13:
@ruralaccent ?

bjoernbusch, 2020-12-15:
I'm facing the same issue on Octoprint 1.5.2 and UI Customizer: 0.1.1.4. In the control tab it's working, but the Webcam on the dashboard just says loading all the time. Any additional info I could give you to figure this out?

LazeMSS, 2020-12-15:
> I'm facing the same issue on Octoprint 1.5.2 and UI Customizer: 0.1.1.4. In the control tab it's working, but the Webcam on the dashboard stops working once I switch to a different tab in Chrome. Any additional info I could give you to figure this out?
> 
> Could this have anything to do with it?
> [image]
> 
> I have Google Chrome Version 87.0.4280.88 (Official Build) (64-bit) in Windows 10.

That bug is fixed - I made an error in the latest version :(

zer0ish, 2020-12-15:
[
ENTER DESCRIPTION HERE- ALSO ADD SCREENSHOT IF POSSIBLE!
Done

Describe your problem?
When I change tabs in Chrome Version 87.0.4280.88 (Official Build) (64-bit) then go back to my Octoprint tab, I get the message Loading Camera, but the camera never loads. This wasn't an issue 2 version ago. Started with the last 2 patches.

What is the problem?
The camera doesn't load.

Can you recreate it?
Yes.

Did you try disabling plugins?
No, this wasn't an issue before the last 2 patches. Before it would say "failed to load camera and stop trying".

Did you remember to update the subject?
Yes
]

Plugins installed:

AutoLogout (This is turned off)
HeaterTimeout
PrintTimeGenius
action_command_notification
action_command_prompt
announcements
backup
bedlevelvisualizer
discovery
errortracking
filamentmanager
gcodeviewer
navbartemp
pi_support
pluginmanager
softwareupdate
themeify
tracking
uicustomizer
virtual_printer
Software versions:

OctoPrint 1.5.2
Python 2.7.16
OctoPi 0.17.0
[image]

fraintt, 2020-12-15:
I have the same issue, strangely in 0.1.1.3 it worked fine without issues, after the Update to 0.1.1.4 it loads fine after octoprint restart, then when you switch tabs or etc. it changes to "loading" and does not load. When I double-click it opens the "bigger-view" running: OctoPrint 1.5.2, Python 3.7.3, OctoPi 0.17.0 on raspberry pi 3b+ with usb webcam, on firefox 83.0 64bit.

LazeMSS, 2020-12-15:
@fraintt / @zer0ish  / @bjoernbusch  / @ruralaccent should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.5 - if not then reopen please 👍

closed 2020-12-15 by LazeMSS

ruralaccent, 2020-12-16:
Seems ok. Many thanks once again!

fraintt, 2020-12-20:
Can confirm, is working. Thank you!