LazeMSS/OctoPrint-UICustomizer #26: Maximized webcam stream doesn't care about browser height.

Akegata, 2020-12-12:
If I have my browser in a 1:1 aspect ratio or less (for example 1:2) the maximized webcam stream is fully visible, although not filling as much of the browser as possible.
When I change the browser size to something like 16:9, the stream gets to big vertically to see all of the picture, and also makes it impossible to minimize the stream since that icon dissapears.
It seems like it's set to always be a specific number of pixels from the left and right sides of the browser since that doesn't change when the browser size is changed, but not to the top and bottom sides.

Akegata, 2020-12-12:
1:1(ish) aspect ratio:
[image]

2:1(ish) aspect ratio:
[image]

LazeMSS, 2020-12-13:
I can see you are using the "FullScreen webcam plugin" - that adds a lot of extra stuff to the bottom of the webcam - my plugin doesn't take that extra stuff into consideration - I will try and look if its possible

LazeMSS, 2020-12-13:
I maybe I can fix - will try and include it in next release

LazeMSS, 2020-12-13:
@Akegata 
Please confirm its fixed in the latest release:
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.3
... if so then feel free to close this - thanks.

Akegata, 2020-12-14:
I just updated to 0.1.1.3 and also removed the Fullscreen Webcam plugin to avoid issues.
It is better now, it works great on normal 16:9 screens, but on my ultrawide 21:9 screen, the issue is still there.

LazeMSS, 2020-12-14:
The issue still being there is the problem with webcam being hiden behind the topbar?

LazeMSS, 2020-12-14:
I have tried to fix it a bit better in the next release

LazeMSS, 2020-12-14:
Please report browser size and webcam sizes

holocronology, 2020-12-14:
Screen: 1920x1080
Browers size (?): full screen
Webcam: PiCam v2.1, I think, 16:9

Is this what you need?

LazeMSS, 2020-12-14:
The output resoultion on that cam should be 1920x1080 i think - so similar to mine.
Can you post a screenshot of what looks wrong?

holocronology, 2020-12-14:
@LazeMSS I did in the one you closed. Here it is again: https://user-images.githubusercontent.com/65265114/102084416-bdb73f00-3dda-11eb-856f-1e43400dc3c0.png

LazeMSS, 2020-12-14:
I meant fullscreen - I cant figure out why the overlay window is higher than the screen - it should never happen - and you are running the latest 0.1.1.3?

Akegata, 2020-12-14:
I don't know what the webcame video size is for me. I guess it's probably 1920x1080 for me a well, but not sure if that matters since it scales with the browser size anyway?
This is what it looks like for me when I have the browser in fullscreen on my monitor with 3440x1440 resolution.
[image]

LazeMSS, 2020-12-14:
Firefox?

Akegata, 2020-12-14:
It's a normal Logitech something webcam.

But you are onto something with the browser. I tested in both Chrome and Edge as well, there the webcam window doesn't extend behind the topbar.

LazeMSS, 2020-12-14:
Firefox is broken with something with the normal webcam streams - I will try more :)

muenznej, 2020-12-14:
just wanted to mention that i got the same problem with the fullscreen webstream not scaling properly.

LazeMSS, 2020-12-14:
0.1.1.4 release out now - should hopefully fix these problems - please test :)

Akegata, 2020-12-14:
Well, it's definitely better since it doesn't go underneath the topbar, which means the minimize button is accessible. However, the resize is still weird, it doesn't care about the lower part of the screen now.

The print bed should be visible, but the lower part of the image is cut.

[image]

LazeMSS, 2020-12-14:
I cant figure out why - there shouldn't be any cache ie in the browser.
This is on my Firefox with a simulated screen similar to yours
[image]

holocronology, 2020-12-14:
Looks like a lot of others have weighed in. Here's the full screen you asked for: 
[image]

LazeMSS, 2020-12-14:
To check for the latest update got into:
Settings -> Software update. At the bottom of the page press "Advanced options" and then " Force check for update (overrides cache used for update checks)"

@Akegata  can you try this aswell?

Akegata, 2020-12-14:
@LazeMSS I'm not sure if I understand what you mean, it looks like your example is cropped in the same way as mine?
The latest picture I posted was with 0.1.1.4.

LazeMSS, 2020-12-14:
yes you are right - I need to get some sleep - will look a bit more tomorrow

LazeMSS, 2020-12-15:
@Akegata / @holocronology   should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.5 - if not then reopen please 👍

closed 2020-12-15 by LazeMSS

Akegata, 2020-12-15:
Yup, works great now! Thanks!

holocronology, 2020-12-15:
Also confirming fixed in FireFox.

muenznej, 2020-12-16:
all fine. gj and thx