LazeMSS/OctoPrint-UICustomizer #27: [BUG] Top of expanded webcam view is behind fixed top navbar

holocronology, 2020-12-12:
[
 When expanding the webcam view to "full screen" the top of it is hidden behind the top fixed navbar. You can drag it down to get to the click point to close/minimize it. 
[image]
]

**Plugins installed:**
- DisplayLayerProgress
- GcodeEditor
- PrintTimeGenius
- action_command_notification
- action_command_prompt
- announcements
- arc_welder
- backup
- bedlevelvisualizer
- consolidatedtabs
- discovery
- errortracking
- filamentmanager
- gcodeviewer
- marlingcodedocumentation
- pi_support
- pluginmanager
- simpleemergencystop
- softwareupdate
- tracking
- uicustomizer
- virtual_printer

**Software versions:**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

LazeMSS, 2020-12-13:
@holocronology 
Please confirm its fixed in the latest release:
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.3
... if so then feel free to close this - thanks.

holocronology, 2020-12-14:
Bug appears to remain.

The top right corner, with the minimize click point, is outside the window. However, the expanded window is now on top of the fixed navbar. Previously, it was behind the fixed navbar.

[image]

LazeMSS, 2020-12-14:
What is the screen/browser size and webcam resoultion you are using?

LazeMSS, 2020-12-14:
Btw. closing this bug - this is fixed - the size problem is also here - please report size etc. there
https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/26

closed 2020-12-14 by LazeMSS