LazeMSS/OctoPrint-UICustomizer #28: Errors in Console

jneilliii, 2020-12-12:
Not sure yet what this might be but noticed the following error in developer console when trying to debug an issue with PlotylTempgraph. I think it may be related to the terminal pausing, but haven't really dug in yet.

```
packed_core.js:965 Error calling onAllBound on view model UICustomizerViewModel : TypeError: Cannot read property 'pause' of undefined
    at UICustomizerViewModel.self.CustomW_initWebCam (http://192.168.0.120/static/webassets/packed_plugins.js?eddab4c5:22608:44)
    at String.<anonymous> (http://192.168.0.120/static/webassets/packed_plugins.js?eddab4c5:22232:68)
    at Function.each (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:2976)
    at S.fn.init.each (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:1454)
    at Array.<anonymous> (http://192.168.0.120/static/webassets/packed_plugins.js?eddab4c5:22218:32)
    at Function.each (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:2976)
    at S.fn.init.each (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:1454)
    at UICustomizerViewModel.self.set_rowsLayout (http://192.168.0.120/static/webassets/packed_plugins.js?eddab4c5:22199:29)
    at UICustomizerViewModel.self.UpdateLayout (http://192.168.0.120/static/webassets/packed_plugins.js?eddab4c5:22136:26)
    at UICustomizerViewModel.self.onAllBound (http://192.168.0.120/static/webassets/packed_plugins.js?eddab4c5:22014:26)
    at callViewModelIf (http://192.168.0.120/static/webassets/packed_core.js?b8ae0794:964:75)
    at http://192.168.0.120/static/webassets/packed_core.js?b8ae0794:958:138
    at Pn (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:94753)
    at Function.<anonymous> (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:105184)
    at callViewModelsIf (http://192.168.0.120/static/webassets/packed_core.js?b8ae0794:958:95)
    at callViewModels (http://192.168.0.120/static/webassets/packed_core.js?b8ae0794:957:56)
    at bindViewModels (http://192.168.0.120/static/webassets/packed_core.js?b8ae0794:1024:87)
    at sentryWrapped (http://192.168.0.120/static/webassets/packed_core.js?b8ae0794:487:45795)
```

jneilliii, 2020-12-12:
Actually appears to be related to the webcam stuff. If the webcam stream url is empty the error occurs, but after putting in the stream url it works without error.

LazeMSS, 2020-12-13:
Thanks - fixed in next release

LazeMSS, 2020-12-13:
@jneilliii 
Please confirm its fixed in the latest release:
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.3
... if so then feel free to close this - thanks.

jneilliii, 2020-12-13:
Confirmed fixed, but might recommend you not even show your sidebar webcam at all if webcam isn't enabled. I know it's easy enough to hide it in the settings, but just something to consider.

closed 2020-12-13 by jneilliii

LazeMSS, 2020-12-14:
I will leave it as - so people turning it on and wondering why there is no webcam will configure it :)