LazeMSS/OctoPrint-UICustomizer #29: Webcam orientation and Gcode Viewer size

JanB97, 2020-12-13:
The Webcam on the side  does not care about the orientation settings in octoprint.
It is always the wrong orientation until I go in and disable and enable to flip the webcam again.
After I reload the webpage it is wrong again.

Also the GCode viewer doesn't scale if the center row is more than 4.

And I have a feature request for this otherwise great Plugin! (I don't know where to post this else)
Could you maybe make like a extra layout category to have buttons that execute own scripts or something like that?
All other options involve GCode which does freeze the print...

LazeMSS, 2020-12-13:
What is the URL of your webcam? Is it HLS or normal?
Post a screenshot of you webcam settings and share what it should look when it works and when it doesn't work.

Everything works fine for me and everybody else testing - so I need more info.

JanB97, 2020-12-13:
Its an IR Camera using the PI Webcam Interface.
The URL is: /webcam/?action=stream

This is how to should look: 
[image]

This is how it does look:
[image]

It seems like this only happens if the main cam is hidden.
It also doesn't seem to happen every time.

Settings:
[image]

This is the Gcode Viewer:
[image]

LazeMSS, 2020-12-14:
The gcode viewer is full width on my installation - it could be one of your plugins changing the way the canvas is done. I will look later on the other stuff.

LazeMSS, 2020-12-14:
I have run my webcam the same way as yours for 4-5 hours now - no problems at my part - i can' reproduce

LazeMSS, 2020-12-14:
@JanB97  try the latest version and see if it still happens. - thanks - if not then please close this issue

JanB97, 2020-12-14:
Okay, you are right, it's the Plugin "Exclude Region" that causes the GCode Viewer behavior in connection with your Plugin.
And the Webcam Problem is caused with the Plugin Dashboard and your Plugin active.

As it is in connection with other Plugins it might not really be a Problem of your Plugin, but it is only happening if yours is active.
As I find your way better in usability, I will however deactivate those and keep yours active. 
Thanks :)

closed 2020-12-14 by JanB97

LazeMSS, 2020-12-14:
I will look and see what causes it with these plugins