LazeMSS/OctoPrint-UICustomizer #30: Enhancement request: Combining some of the main tabs

DrewBatchelor, 2020-12-13:
This plug in is brilliant, Thank you.
Just an idea for an enhancement request, It would be amazing if it could be possible to combine the main tabs.
For example the Temperature tab and the Dashboard tab. With the temperature tab on top and Dashboard tab below would fit on 1 screen for me.
Likewise Terminal and Klipper could be stacked into 1 tab as well. 
Kind Regards
Drew

jneilliii, 2020-12-23:
You could technically use Consolidated Tabs plugin in conjunction with UI Customizer to achieve this.

[image]

Jack1337, 2020-12-23:
Really thank you for this plugin! its awesome!
it would be awesome when we could combine tabs like klipper and terminal, or Temperatur and Control.
Or maybe not combine them, just split the MainTabs and place one by one.
so i cloud place both among themselves.

Thank you!!

LazeMSS, 2020-12-24:
> Really thank you for this plugin! its awesome!
> it would be awesome when we could combine tabs like klipper and terminal, or Temperatur and Control.
> Or maybe not combine them, just split the MainTabs and place one by one.
> so i cloud place both among themselves.
> 
> Thank you!!

As @jneilliii  pointed out he already has a plugin that does some of this

jneilliii, 2020-12-24:
I'd also be happy to collaborate on making that plugin more compatible with UI Customizer.

LazeMSS, 2020-12-25:
> I'd also be happy to collaborate on making that plugin more compatible with UI Customizer.

When back from holiday I will probally make some changes :)

NovaViper, 2021-02-24:
Is there any progress on this? I'd be more than willing to test this out on my Octo instance

jneilliii, 2021-02-24:
I've made some progress on my side. If you want to try it out, you can install by copying/pasting the URL below into plugin manager > get more > ...from URL. Sizing the panels seems way more stable now since implementing gridstack and @LazeMSS has helped with getting through an initial bug I was having with the implementation. I've had positive feedback so far with this version on its own and it is compatible with UI Customizer now.

```
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip
```

NovaViper, 2021-02-24:
@jneilliii Er.. is it suppose to show only the settings? Im only getting this and no option to modify it
[image]
[image]

jneilliii, 2021-02-24:
click the edit button (pencil icon) in the navbar...

NovaViper, 2021-02-24:
Ah I see! Thanks, I got it together and it seems to be working fine with the control, temp and OctoKlipper tabs.

David-Maisonave, 2021-04-14:
I didn't know there was a ConsolidatedTabs Plugin.  I just tried both together and they work great. It would be nice if the two plugins could be consolidated into one plugin.

Thank you both for these great plugins.

LazeMSS, 2022-05-21:
Closing - handled by ConsolidatedTabs

closed 2022-05-21 by LazeMSS