LazeMSS/OctoPrint-UICustomizer #31: Slow/Hang on Firefox

Metathesus, 2020-12-13:
First, Have to say it's a great plugin but, it is extremely slow/ it hangs with Firefox but no issues with Chrome!

My webcam stream is another Octoprint stream.
Other plugins that I have :
- Bed Visualizer (1.0.0)
- Convert TF to SD (1.0.0)Convert TF to SD in printer responses
- Custom CSS (0.0.1)
- Dashboard (1.16.3) 
- DisplayLayerProgress Plugin (1.24.0)
- FileManager (0.1.4)
- GcodeEditor (0.2.10)
- SpoolManager Plugin (1.2.0)
- Themeify (1.2.2)

My webcam stream is another Octoprint stream.

LazeMSS, 2020-12-13:
And you don't have any problems when my plugin is disabled?
Have you tried disabling the webcam stream totally?

Metathesus, 2020-12-14:
Yes this problem doesn't occur with your plugin disabled of course.

I've missed to say when Chrome loading it says "!error loading webcam" in the webcam panel but the stream load fine in the control tabs.
[image]
And after seconds or minute the videostream appear in your webcam panel.

**If I disable webcam, the panel show "loading webcam" and Octoprint get back is fast behavior.**

LazeMSS, 2020-12-14:
What is the URL your are using for webcam?

Metathesus, 2020-12-14:
My octoprint 1.5.1 IP is: 192.168.1.45
Another Octoprint 1.4.x that deliver the stream :
http://192.168.1.40:8080/?action=stream

LazeMSS, 2020-12-14:
Well is extremely slow in my firefox aswell even when I have UI Customizer disabled - but if you say you have no problems then it might be my installation. I will try and look further.

Are you running windows 10?

LazeMSS, 2020-12-14:
@Metathesus  please try the latest version: 0.1.1.4 - I found a bug :(

Metathesus, 2020-12-15:
Seems to be far better (after CTRL+F5 to clear cache)!

Thanks for your fix and all the work you do!

______________
As a suggestion for the file wrapper to better be handled with your plugin, when the file wrapper is in a large central column, I use this css (far from optimized but enough for the moment) with plugin CustomCSS
div.internal{display:none;}
div.uploaded{float:right; margin-top: -15px; margin-right:140px; width:140px;}
div.size{float:right; margin-top:-15px; margin-right: 50px;/*display:none*/}
div.toggleAdditionalData{float:left; margin-right:10px;}
div.scroll-wrapper{height:auto !important;}

You can probably do something to get the same result but in a better way! ;-)

LazeMSS, 2020-12-15:
Thanks I will look into it asap :)

closed 2020-12-15 by LazeMSS