LazeMSS/OctoPrint-UICustomizer #32: DragonOrder & Dashboard Issues

Pheelix, 2020-12-14:
DragonOrder:
Can't move some Navbar Icons.  I have my navbar set to centered in UI Customizer.  The ones I can move are on the right of the navbar.  But there are 2 that are centered with Navbar Temp that I can not move (PSU Control, Enclosure, and Simple Emergency Stop).  There could be others as well that have add to navbar feature.  When I try to move these two, it says upload at the bottom of the screen and Javascript void by the mouse cursor.

[image]

Dashboard:
When going into Fullscreen or Full Browser.  UI Customizer Overlays on top of the Dashboard.  I don't know if you will need to contact the creator of Dashboard to fix this.  Or add a option for the UI to be hidden when Dashboard is in Fullscreen or Full Browser mode.

[image]

Suggestion for Layouts:
Able to make the Layout Row Size more customizable.  example would be a box you can enter a number in to set row width px size.  Was able to do this with themify without UIC.

LazeMSS, 2020-12-14:
Basically DragonOrder and UI Customizer are not compatible and its hard to make them - I haven't looked into dashboard yet.

The widths are set using rows - I don't want to add pixel based sizes - what your are looking could probally me made using: https://plugins.octoprint.org/plugins/customcss/

j7126, 2020-12-14:
Hi, dashboard maintainer here, I have just installed the UI customizer plugin, and dashboard full screen seems to be working fine. 

I did experience an issue where with the UI customizer plugin installed, dashboard would not open full browser mode, however that does not seem to be the same issue here and I have a fix ready for the next release of dashboard. 

However I cannot seem to reproduce the issue shown in your screenshot. @Pheelix can you please try disabling other plugins that you have installed other than dashboard and UI customizer so we can confirm that the issue is only between these two plugins.

LazeMSS, 2020-12-14:
@j7126 thanks for dropping by :) I just tried you plugin (awesome btw.) - I could not recreate these problems.

My guess is DragonOrder should be disabled and a test.

Pheelix, 2020-12-14:
I have all my plugins but UICustomizer, Dashboard, Displaylayer Progress, and Custom CSS (no code in box) turned off for testing to see what was causing the issues. 

I found the issue with the Dashboard Full Screen Overlay.  It was the Use theme colors in fullscreen mode in Dashboard > Other settings.
[image]

And noticed another issue.  The background is all black and not matching the non fullscreen color.  As you can see what color the background is suppose to be from my first post.
[image]

As far as DragonOrder.  The only issue I am having with it is that I can not move the Navbar icons bye the Navbar Temp.  As I said in the first post I am only able to move the order of the ones on the right and they will only stay to the right.  Also made sure any css modifications in themify are not enabled as I did use some to adjust the Navbar before UIC.  I disabled those once UIC was installed and they were removed before these issues.

The only difference I notice about the Navbar Icons that can be moved is that they are Icons that Octoprint has up there by default.  And the 2 by the Navbar Temp are 3rd Party Plugin ones that can be added from there plugings settings.  Maybe the way the Icons are coded to be put on the Navbar could have something to do with it?

If you need me to test anything else just let me know.

Pheelix, 2020-12-14:
@LazeMSS would it be possible to add some containers in the navbar for the 3rd party plugin navbar icons.   This way we can sort them in the UIC settings, maybe something like what you did with the Tab Icons+text.

jneilliii, 2020-12-14:
The ordering that DragonOrder does is purely a UI hack to allow saving the desired order into octoprint's config.yaml file. The ordering itself is handled by octoprint and the plugin is not necessary to control that.

Pheelix, 2020-12-14:
@jneilliii ok that is good to know.  Also about the black background from turning off use theme in full screen.  I went into developer mode on my browser to find the div id so I could have themify change it.  when I unchecked background-color in .dashboardContainer.dashboard-full from the styles.  I got the same issue as my screenshot in the first post.

I checked the errors in the developer mode and found this: 

> Refused to apply style from 'http://octopi.local/static/webassets/packed_plugins.less?d4c89092' because its MIME type ('application/octet-stream') is not a supported stylesheet MIME type, and strict MIME checking is enabled.

and

> DevTools failed to load SourceMap: Could not load content for http://octopi.local/static/webassets/bootstrap-colorpicker.css.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE

Not sure if this helps at all.

jneilliii, 2020-12-14:
For details on ordering check the order section of the appearance config [here](https://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance). I don't know if that's how @LazeMSS is doing his re-ordering or not, but that's how all my ordering plugins work, just saving to those global settings.

LazeMSS, 2020-12-14:
@jneilliii  great thanks - didn't know about these settings :)

LazeMSS, 2020-12-14:
> @LazeMSS would it be possible to add some containers in the navbar for the 3rd party plugin navbar icons. This way we can sort them in the UIC settings, maybe something like what you did with the Tab Icons+text.

Its in the roadmap - for now I will close this issue if okay - if not then reopen :)

closed 2020-12-14 by LazeMSS

jneilliii, 2020-12-14:
> @jneilliii great thanks - didn't know about these settings :)
> 
> Updated: I don't think I will use this - although more "standard" way it will be hard to cleanup at uninstall etc.

That's understandable. I used it out of convenience and not have to do startup processing within the plugin itself. There's lots in config.yaml that people don't know about. For example custom controls, and system commands.