LazeMSS/OctoPrint-UICustomizer #33: 1.1.3 Installed and Octolapse no more working

pierremilcent, 2020-12-14:
I just istalled 1.1.3 and got a probleme with Octolapse now :+1: 
2020-12-14 14:11:58,951 - octolapse.__init__ - ERROR - Unable to start the timelapse.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 2427, in on_print_start
    timelapse_settings = self.get_timelapse_settings()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 2666, in get_timelapse_settings
    success, error_type, error_list = current_printer_clone.get_gcode_settings_from_file(gcode_file_path)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/settings.py", line 682, in get_gcode_settings_from_file
    new_slicer_settings.update_settings_from_gcode(new_settings, self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/settings.py", line 2789, in update_settings_from_gcode
    num_extruders = int(settings_dict["machine_extruder_count"])
TypeError: int() argument must be a string or a number, not 'NoneType'

LazeMSS, 2020-12-14:
That is 100% not a problem with my plugin. My guess is that your setup currently doesn't report any extruders and thats why its throwing that fault.

closed 2020-12-14 by LazeMSS

pierremilcent, 2020-12-14:
I found my error, i tried to do a timelapse with a part of one layer only (bed leveling), just after i updated your plugin. Closed.