LazeMSS/OctoPrint-UICustomizer #34: Drop-down menu hidden behind main tabs

Merhim, 2020-12-14:
Hi,
The central drop-down menu in the top-bar are hidden by the main tabs, so they are difficult to reach.
Thanks for your work
 
[image]

LazeMSS, 2020-12-14:
There seems to be a problem with something overlaying the mains tabs - maybe another plugin. 

But I need more information - what other plugins are you using?
What settings in my plugin?
And what version of my plugin are you using?

LazeMSS, 2020-12-14:
From my installation: 
[image]

Merhim, 2020-12-14:
Everything works when "Fixed header/topbar" is selected or "Center topbar icons" is not selected.
The problem remains in the configuration below

UI Customizer version: 0.1.1.4 

[image]

LazeMSS, 2020-12-15:
Found the bug - will fix it in the next release

LazeMSS, 2020-12-15:
@Merhim should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.5 - if not then reopen please 👍

Merhim, 2020-12-20:
Hi Laze,
I have just updated UICustomizer to the new v0.1.1.6, but i still have the same bug as before.

[image]
[image]

Merhim, 2020-12-20:
As before the problem only occurs when the header/topbar is not fixed.

Thank you so much for you effort  👍

LazeMSS, 2020-12-21:
Should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.7
Please try and if not fixed then reopen this issue.

closed 2020-12-21 by LazeMSS