LazeMSS/OctoPrint-UICustomizer #37: Loading webcam… (ver. 0.1.1.4)

apainter2, 2020-12-15:
[
Webcam frame works for a few minutes then times out, and says 'Loading webcam' (see screenshot)
 Describe your problem? - As above
 What is the problem? - As above
 Can you recreate it? - Yes
 Did you try disabling plugins? - Yes - Problem goes away as webcam no longer is shown as UI Customizer is not loaded
 Did you remeber to update the subject? - yes
]
[image]

**Plugins installed:**
Bed Visualizer - 1.0.0
Creality Temerature - 1.2.4
Creality-2x-temperature-reporting-fix - 0.0.4
Cura Thumbnails - 0.2.1
DisplayLayerProgress Plugin - 1.24.0
Filament Manager - 1.6.3
HeaterTimeout - 0.0.3
Navbar Temperature Plugin - 0.14
Octolapse - 0.4.1
Preheat Button - 0.5.1
PrintTimeGenius - 2.2.7
Resource Manager - 0.2..6
UI Customizer - 0.1.1.4

Pi Support Plugin
Virtual Printer

**Software versions:**
- OctoPrint 1.5.2
- Python 3.7.3
- OctoPi 0.17.0

LazeMSS, 2020-12-15:
its fixed in the latest version 0.1.1.5 - please try and update.

apainter2, 2020-12-15:
I dont have 0.1.1.5 listed as an upgrade within OctoPrint

apainter2, 2020-12-16:
Upgraded to 0.1.1.5 - Thank you.

closed 2020-12-16 by apainter2