LazeMSS/OctoPrint-UICustomizer #40: Interface totally gone !!!

Lecl0ne, 2020-12-22:
## Description
Interface totally disappear except header and footer.
Can you recreate it? I can't get normal display back with plugin activated.
Did you try disabling plugins? Yes. It solves the problem. But re-enabling it makes the problem back. So for now plugin is disabled...
<hr>

**Plugins installed**
- Active Filters Extended[active_filters_extended] v0.1.0
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- Autologin Configuration[autologin_config] v0.2.0
- Autoscroll[autoscroll] v0.0.3
- Autoselect Plugin[autoselect] v0.3.0
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Cancel Objects[cancelobject] v0.4.4
- Cost Estimation[costestimation] v3.1.0
- Detailed Progress[detailedprogress] v0.2.6
- Friendly Neighborhood Beeper[FriendlyBeeper] v0.4.0
- Fullscreen Plugin[fullscreen] v0.0.6
- GCODE System Commands[gcodesystemcommands] v1.0.1
- M117NavBar[M117NavBar] v0.1.1
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- Multi Line Terminal[multilineterminal] v0.1.5
- Octolapse[octolapse] v0.4.1
- Preheat Button[preheat] v0.5.1
- Printoid Plugin[printoid] v1.0.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Progress Title[progress_title] v0.1.1
- ProgressBasedOntime Plugin[ProgressBasedOnTime] v0.1.6
- SetAndWait[setandwait] v0.0.1
- SpoolManager Plugin[SpoolManager] v1.2.0
- Terminal Messaging[terminalmessaging] v0.1.0
- Themeify[themeify] v1.2.2
- **UI Customizer[uicustomizer] v0.1.1.7**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: true
- fixedFooter: false
- fixedHeader: false
- fluidLayout: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: false

**Software versions**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36 Edg/87.0.664.66

LazeMSS, 2020-12-22:
I can't recreate that bug - have you tried disabling other 3rd party plugins?

What does the developer console say?

Lecl0ne, 2020-12-22:
I don't know what 3rd party plugin to disable, everything worked flawlessly yesterday.
just turned on "Improve mobile/responsive", same.

[image]

Lecl0ne, 2020-12-22:
Another very strange thing. Since this bug, look at the Layout tab. It's all empty and impossible to raise the right row width (forbiden mouse icon)

[image]

LazeMSS, 2020-12-22:
Try turning of 3rd party plugins - i would suggest themify first

Lecl0ne, 2020-12-22:
Just tried. Same. :/

LazeMSS, 2020-12-22:
Do you have an option for running teamviewer/etc. so we can inspect together?

LazeMSS, 2020-12-22:
drop me an email at:
uicust 
at
jemi
.
dk

remove newlines

Lecl0ne, 2020-12-22:
Mmmm i'm at work right now, no Teamviewer installed.. At home no problem sir.
I just cleaned UICust. datas and everything is back again. I'm setting the plugin again to see if the issue is back.

LazeMSS, 2020-12-22:
Strange - maybe the settings got corrupted. Let me know if thats the problem.

LazeMSS, 2020-12-23:
Did you get it fixed?

Lecl0ne, 2020-12-23:
Yep, no problem so far...

LazeMSS, 2020-12-23:
Okay closing

closed 2020-12-23 by LazeMSS

MetalTurtle18, 2020-12-27:
Hi, I hate to comment on a closed issue but this exact same thing just happened to me.

LazeMSS, 2020-12-27:
> Hi, I hate to comment on a closed issue but this exact same thing just happened to me.

Im looking into it - but the only reason I can think off it because a config file gets corrupted - what does your browsers developer console say?

Ajimaru, 2020-12-29:
I have the same issue, the UI become corrupted for no reason. disable and enable UI customizer plugin does not fix the issue.
Reinstall the UI customizer plugin resolves the issue. Other Ui related plugins are deactivated.

[image]

LazeMSS, 2020-12-30:
I will look. Can you look in your settings or maybe show me the developer console when it happens

MetalTurtle18, 2020-12-30:
I already reinstalled it but if it happens again I will be sure to do this.

Ajimaru, 2020-12-30:
@LazeMSS which settings do you mean?

LazeMSS, 2020-12-30:
I meant any settings in my plugin..but I would need a developer console output or Teamviewer etc to inspect what is happening