LazeMSS/OctoPrint-UICustomizer #41: M117 area: texte is below

Lecl0ne, 2020-12-22:
## Description
M117 Navbar plugin: text is below the area
Can you recreate it? Sometime the M117 messages are in the area, sometime not. Not depending on the message length. Refreshing the page is sometime anough to solve it.
Did you try disabling plugins? Yes. It solves the problem.
Did you remember to update the subject? Course
<hr>

**Plugins installed**
- Active Filters Extended[active_filters_extended] v0.1.0
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- Autologin Configuration[autologin_config] v0.2.0
- Autoscroll[autoscroll] v0.0.3
- Autoselect Plugin[autoselect] v0.3.0
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Cancel Objects[cancelobject] v0.4.4
- Cost Estimation[costestimation] v3.1.0
- Detailed Progress[detailedprogress] v0.2.6
- Friendly Neighborhood Beeper[FriendlyBeeper] v0.4.0
- Fullscreen Plugin[fullscreen] v0.0.6
- GCODE System Commands[gcodesystemcommands] v1.0.1
- M117NavBar[M117NavBar] v0.1.1
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- Multi Line Terminal[multilineterminal] v0.1.5
- Octolapse[octolapse] v0.4.1
- Preheat Button[preheat] v0.5.1
- Printoid Plugin[printoid] v1.0.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Progress Title[progress_title] v0.1.1
- ProgressBasedOntime Plugin[ProgressBasedOnTime] v0.1.6
- SetAndWait[setandwait] v0.0.1
- SpoolManager Plugin[SpoolManager] v1.2.0
- Terminal Messaging[terminalmessaging] v0.1.0
- Themeify[themeify] v1.2.2
- **UI Customizer[uicustomizer] v0.1.1.7**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: false
- disableTermInactive: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: false

**Software versions**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36 Edg/87.0.664.66

LazeMSS, 2020-12-22:
Screenshots?

LazeMSS, 2020-12-22:
Bug found :)

Lecl0ne, 2020-12-22:
Well done ! It was not possible to make a screenshot cause to my other issue :)

Lecl0ne, 2020-12-23:
Here is the screenshot !

[image]

LazeMSS, 2020-12-23:
Should be fixed now:
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.8

Please reopen if not fixed.

closed 2020-12-23 by LazeMSS