LazeMSS/OctoPrint-UICustomizer #43: Gcode viewer doesn't fit the dedicated area

Lecl0ne, 2020-12-24:
In a column width of 5 (and maybe 4 too btw), the G-code viewer is not optimized and is smaller than it can be, see screenshot

[image]

Can you recreate it? It's column width dependant.

<hr>

**Plugins installed**
- ProgressBasedOntime Plugin[ProgressBasedOnTime] v0.1.6
- Friendly Neighborhood Beeper[FriendlyBeeper] v0.4.0
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- Fullscreen Plugin[fullscreen] v0.0.6
- Cancel Objects[cancelobject] v0.4.4
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- Themeify[themeify] v1.2.2
- Autoscroll[autoscroll] v0.0.3
- Terminal Messaging[terminalmessaging] v0.1.0
- **UI Customizer[uicustomizer] v0.1.1.8**
- Autoselect Plugin[autoselect] v0.3.0
- Autologin Configuration[autologin_config] v0.2.0
- M117NavBar[M117NavBar] v0.1.1
- Printoid Plugin[printoid] v1.0.2
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- SetAndWait[setandwait] v0.0.1
- Progress Title[progress_title] v0.1.1
- Detailed Progress[detailedprogress] v0.2.6
- GCODE System Commands[gcodesystemcommands] v1.0.1
- Cura Thumbnails[UltimakerFormatPackage] v0.2.1
- Preheat Button[preheat] v0.5.1
- Multi Line Terminal[multilineterminal] v0.1.5
- SpoolManager Plugin[SpoolManager] v1.2.0
- Cost Estimation[costestimation] v3.1.0
- Active Filters Extended[active_filters_extended] v0.1.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Octolapse[octolapse] v0.4.1

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: false
- disableTermInactive: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: false

**Software versions**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36 Edg/87.0.664.66

LazeMSS, 2020-12-24:
Yes I stepped back from using full width due to the aspect ratio of the gcode causing the controls to slip to the bottom. I might add it as a feature.

Lecl0ne, 2020-12-24:
> causing the controls to slip to the bottom.

Can you show the problem in a screenshot ?

LazeMSS, 2021-01-01:
Fixed in 0.1.1.9(https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.9)
- please check and reopen/comment if not working as expected.

Lecl0ne, 2021-01-04:
Hey Laze,

Problem still here for me after enabling your new option:

[image]

LazeMSS, 2021-01-04:
Tried refreshing? it works fine - I haven't tried your layout but you can check using the developer tools, inspector if you know how

Lecl0ne, 2021-01-05:
I tried refreshing, wiping cache etc.. I dont know about inspector :(

LazeMSS, 2021-01-05:
Right click on the gcode viewer and select inspect. It will show you a lot info take a screenshot of that

LazeMSS, 2021-01-05:
I have tried on all my computers and devices. In the next version the preview will show when its applied. I noticed you are runinng 0.1.18 on the bug report and using the fullscreen plugin. There might be problems.

Anyway after next release please try again.

Lecl0ne, 2021-01-06:
Still the same. You're right. It's about "Fluid/full width layout" option. When enabled, it's not ok...

LazeMSS, 2021-01-06:
Can you take a screenshot view preview on and toggling it on/off - here how it looks on my pc:

Without:
[image]

With:
[image]

LazeMSS, 2021-01-08:
@Lecl0ne ?

Lecl0ne, 2021-01-08:
Yep sorry I was busy. I'll do it.

Lecl0ne, 2021-01-08:
Here it is:

Full width On / G-code fullsize On
[image]

Full width On / G-code fullsize Off
[image]

Full width Off / G-code fullsize On
[image]

Full width Off / G-code fullsize Off
[image]

LazeMSS, 2021-01-08:
Have you tried different layouts ? Ie tried having the gcode in the center like mine - just to rule that out?

But my guess is that another plugin is installed that also changes in the way the gcode viewer is handled

Lecl0ne, 2021-01-08:
I'll give it a try.. :/

LazeMSS, 2021-01-08:
Ahh FOUND it - you have disabled responsive mode

LazeMSS, 2021-01-08:
Waiting to close after release

Lecl0ne, 2021-01-09:
Nope.
[image]

LazeMSS, 2021-01-09:
So still not working when you have responsive mode on?
From your original post it looked like you didn't have that on - and I made an error on the full widt only work with this.

Lecl0ne, 2021-01-09:
Screenshots are with responsive mode ON.

LazeMSS, 2021-01-09:
Then I have no idea. Judging from your screenshots https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/43#issuecomment-756966159

The difference in sizes show that symptoms based on improved mobile responsive disable

LazeMSS, 2021-01-09:
Closing - should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.2.4

@Lecl0ne  if not fixed then please reopen this and post screenshots as before

closed 2021-01-09 by LazeMSS

Lecl0ne, 2021-01-12:
Hey,

It's still exactly the same since the update.

LazeMSS, 2021-01-12:
I have no idea why this is happening

Lecl0ne, 2021-01-12:
Is there any plugin I should try to disable ?

Lecl0ne, 2021-01-12:
Something interesting is that the size of the area changes with the Chrome explorer zoom:

100%
[image]

50%
[image]

125%
[image]

Lecl0ne, 2021-01-12:
Found it !!!! Compatibility problem with Cancel Object Plugin !

LazeMSS, 2021-01-12:
I was wondering if that plugin changes anything to the canvas

Lecl0ne, 2021-01-19:
Bug still here with new update.

LazeMSS, 2021-01-19:
I have not tried with the cancel object plugin

LazeMSS, 2021-01-24:
Fixed in next version and supporting cancel object :)

Lecl0ne, 2021-01-24:
Well done!!!

codefaux, 2022-06-13:
Hey so I found this trying to find what was causing a similar issue with a -different- layout plugin. This plugin looks way better though lol -- is this fixed or nah?

LazeMSS, 2022-06-15:
This should be fixed yes

commit 47c2243e (2021-01-08): Fixed #43 and removed remains of less