LazeMSS/OctoPrint-UICustomizer #46: Request - Export/Import

PGlyre, 2020-12-28:
Export / Import function. 
It would have been easier for those who have several printers with OctoPrint and in the first place I think of configuration under "Main Tabs" #
[image]

LazeMSS, 2020-12-28:
You want an option to export the UI settings? That should not be that difficult.

PGlyre, 2020-12-30:
Hi, I'm not the best at English, but you got me right. It was my wish and if you ask if you do not get an answer ....... I have full respect for developers like you and I do not have that kind of knowledge so therefore I submitted this proposal to you

LazeMSS, 2021-11-08:
Just a quick update - im working on this - its not as easy as it might seem but I will fix it :)

jaimieb29, 2021-11-08:
No problem, really appreciate your work thanks

LazeMSS, 2021-11-08:
90% done ;)

LazeMSS, 2021-11-13:
Should now be included in the 0.1.8.0 release (https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.8.0) - please reopen if not working as expected.

closed 2021-11-13 by LazeMSS