LazeMSS/OctoPrint-UICustomizer #47: Interface crashes randomly

Ajimaru, 2020-12-30:
Already discussed in tread [https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/40](url)
I have another crash today while I was adding two macros with the Macro plugin.
The crashes occure also during regular use of the Octoprint UI.
I have attached a copy of the Firefox web-console.
[console-export-2020-12-30_21-7-30.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5755575/console-export-2020-12-30_21-7-30.txt)

LazeMSS, 2020-12-30:
I will look ASAP

LazeMSS, 2021-01-01:
Are you using Klipper plugin?

LazeMSS, 2021-01-01:
Fixed in 0.1.1.9(https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.9)
- please check and reopen/comment if not working as expected.

closed 2021-01-01 by LazeMSS

Ajimaru, 2021-01-04:
The issue was caused by the macro plugin and the timelap tab.
It is working as expected now, thank you.