LazeMSS/OctoPrint-UICustomizer #48: UI goes blank when saving settings

64bittuning, 2020-12-30:
## Description
I recently started using Klipper along with this addon. Each time I entered the Klipper config window to make an adjustment, my entire UI would go blank, with the exception of the top menu bar. I was unable to reinitialize any of the window selections through the plugin. Using the "Clean Plugin Data" from the plugin manager will restore the UI, but ONLY if I do not close out from the window. I must choose "Clean Data" and then "Restart Now" from the Plugin Manager UI notification, otherwise the UI stays blank and I must redo the process.
The blank UI is persistent through Octoprint and System restarts.
I can recreate the plugin every single time.

I have removed old and unused plugins, and gone back to systematically disable everything one at a time. The fault seems to be between your plugin and OctoKlipper.
<hr>

**Plugins installed**
- Terminal Commands Extended[terminalcommandsextended] v0.1.6
- Action Commands[actioncommands] v0.3
- Fullscreen Plugin[fullscreen] v0.0.6
- Gcodebar Plugin[gcodebar] v0.1.5
- OctoKlipper[klipper] v0.3.2
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Themeify[themeify] v1.2.2
- Terminal Response[terminalresponse] v0.1.2
- **UI Customizer[uicustomizer] v0.1.1.8**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36

LazeMSS, 2021-01-01:
Fixed in 0.1.1.9(https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.1.9)
- please check and reopen/comment if not working as expected.

closed 2021-01-01 by LazeMSS

64bittuning, 2021-01-01:
Updated and tried several times to recreate the failure. 
Cannot duplicate,

Thank you very much.

Solandus, 2021-01-02:
I'm sorry to tell you, that this should be reopened. After updating UICustomizer to 0.1.2 everything worked fine until i change some spool data within the FilamentManager Plugin. 

The Interface is gone. Top Bar was left. Deactivation/Reactivation of the UICustomizer plugin doesn't helped. Uninstallation with clearing data and reinstallation helped. I think there something happening within the saved data for the plugin.

Tried to reproduce it, after reinstallation solved it. But no success. 

After both issues looking the same, I have to mentioning https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/40

<hr>

**Plugins installed**
- Python 3 Check[python3plugincompatibilitycheck] v0.1.5
- WS281x LED Status[ws281x_led_status] v0.6.0
- HeaterTimeout[HeaterTimeout] v0.0.1
- Emergency Stop Button[estop] v0.1.2
- BLTouch Plugin[BLTouch] v0.3.4
- Enclosure Plugin[enclosure] v4.13.1
- Cancel Objects[cancelobject] v0.4.4
- MQTT[mqtt] v0.8.7
- Bed Leveling Wizard[bedlevelingwizard] v0.2.4
- Themeify[themeify] v1.2.2
- Autoscroll[autoscroll] v0.0.3
- **UI Customizer[uicustomizer] v0.1.2.0**
- Tab Order[taborder] v0.5.12
- Fan Speed Control[fanspeedslider] v0.2.1
- WebcamTab[webcamtab] v0.2.0
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Detailed Progress[detailedprogress] v0.2.6
- Printoid Plugin[printoid] v1.0.2
- DeleteAfterPrint Plugin[DeleteAfterPrint] v1.8.1
- Extra Distance Buttons[extradistance] v0.1.0
- TP-Link Smartplug[tplinksmartplug] v0.9.26
- Backup Scheduler[backupscheduler] v0.0.4
- Navbar Temperature Plugin[navbartemp] v0.14
- Change Filament Plugin[Change_Filament] v0.2.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Filament Manager[filamentmanager] v1.6.3
- Octolapse[octolapse] v0.4.1
- Firmware Updater[firmwareupdater] v1.7.6

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2 (yes, i know, i have to update this)
- Python 2.7.13
- OctoPi 0.15.1

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36

LazeMSS, 2021-01-02:
@Solandus  If my plugin is disabled then nothing in the UI should be changed by my system - all the changes to the UI happens inside my code.

But you are saying if you change something in Filamament plugin then this still happens - and that happend after you installed the updated version.

Solandus, 2021-01-02:
Nah. Your plugin was activated. With UICustomizer deactivated everything was fine. Reactivating brought back the missing content area. Top and bottom bar was shown.

And yes, that happend after I had updated to 0.1.2 AND changed some settings in sayed Filament Manager Plugin. 

I too saw that within your plugin configuration the layout tab only showed three on the left column. Every other column was clear. And the right column showed a width of 0. (no screenshot, im sry)

As mentioned uninstallation with data clearing and reinstallation solved the problem, there might be some problem with the saved data.

I should also mention that the WebCam Tab plugin moving the webcam into its own tab which I let hide from UICustomizer. And im using themefy. So there may some collisions you already tried to solve.

64bittuning, 2021-01-02:
Perhaps make your own bug report then?

LazeMSS, 2021-01-02:
Yearh these issues might not be related.