LazeMSS/OctoPrint-UICustomizer #49: Pulldown menu in top bar only works 1 times

diedder, 2021-01-01:
## Description
The pulldown menu in the topbar only work at the first click. After clicking one entry the menu on the shutdown or other pulldowns doesn't open.
Steps to recreate: 
- reload page
- clicking on shutdown - pulldown menu opens
- clicking on settings an close the settings window again
- **pulldown menus in the top bar doesn't open anymore**

I was abled to fix this issue by disabling the general option "Fixed header/topbar". 

<hr>

**Plugins installed**
- Resource Monitor[resource_monitor] v0.2.7
- Telegram Notifications[telegram] v1.6.2
- MultiCam[multicam] v0.2.8
- Upload Anything[uploadanything] v1.0.1
- WS281x LED Status[ws281x_led_status] v0.6.0
- Prusa MMU2 Select Filament[mmu2filamentselect] v1.0.2
- Enclosure Plugin[enclosure] v4.13.1
- SlicerSettingsParser[SlicerSettingsParser] v2.0.1
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.3
- **UI Customizer[uicustomizer] v0.1.1.8**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: false
- fluidLayout: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0.2 Safari/605.1.15

LazeMSS, 2021-01-01:
I cant recreated - just tried in all my browsers on both mac, pc and android - can you try a bit more?

madmarf, 2021-01-01:
I can confirm, I got the same problem. PulldownMenus work till I open the configure- or notification-menus. When I try to open the pulldownmenus, it seem like they open behind the UI. I can fix it with refreshing the webpage.

macOS 11.1, Safari 14.0.2, OctoPrint 1.5.2, OctoPi 0.17.0

[image]

diedder, 2021-01-01:
I only use my MacBookPro with Safari. A site refresh fixes it for the next click only. with the option turned off it works stable.

LazeMSS, 2021-01-01:
Please test with the latest version, https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.2.0.
Just tested on my macbook with safari and I cant recreated it.

diedder, 2021-01-02:
after updating to the latest version the behavior is still the same. 
[image]
unfortunately my workaround by deactivating the fixed topbar doesn't work anymore...

I tried also with private surfing without browser cache.

LazeMSS, 2021-01-02:
Can you try and test using chrome on your mac? Is the problem also there?

Is there any custome themes or css applied?

I noticed the black colors but no metion of Themify plugin etc?

diedder, 2021-01-02:
tested on chrome for macos, everything works fine.

There are no themes or css applied, only the "dark mode" of octoprint itself. I noticed another rendering issue in the bottom bar:
[image]

LazeMSS, 2021-01-03:
Yes i see that now - will look into it

LazeMSS, 2021-01-03:
Still can figure out whats causing the dropdown not working. Can you try and record a video of what you do to recreate it - I must be missing something.

diedder, 2021-01-03:
> Still can figure out whats causing the dropdown not working. Can you try and record a video of what you do to recreate it - I must be missing something.

Here a video of both issue with die top/bottom bars
https://www.dropbox.com/s/yf859dumkehfoiu/Bildschirmvideo%20aufnehmen%202021-01-03%20um%2020.51.51.mov?dl=0

LazeMSS, 2021-01-03:
Great thanks - can you try and disabled any other plugins just to see if there is any collission

LazeMSS, 2021-01-03:
Btw. there is a new version out that fixes the bottom issues

diedder, 2021-01-04:
> Great thanks - can you try and disabled any other plugins just to see if there is any collission

the new update fixes my bottom bar.

There seems to be some context with the following plugins:
- WS281x LED Status[ws281x_led_status] v0.6.0
- Enclosure Plugin[enclosure] v4.13.1
if one of them (or both) is activated the pulldown is not working with the option for the topbar true. If both were deactivated it works even with the option activated. You'll need to know, that there plugins creating items in the topbar
[image]

LazeMSS, 2021-01-04:
I cant test LED status plugin :/ I thinks its related to that plugin.

LazeMSS, 2021-01-05:
I cant recreate it anyway. I would probably need a bit of way seeing your installation. Is that possible

diedder, 2021-01-05:
> I cant recreate it anyway. I would probably need a bit of way seeing your installation. Is that possible

Whats‘s your proposal? Maybe we should talk in private about that...

LazeMSS, 2021-01-05:
> > I cant recreate it anyway. I would probably need a bit of way seeing your installation. Is that possible
> 
> Whats‘s your proposal? Maybe we should talk in private about that...

Send me and email at:
uic 
@
jemi
.
dk

remove linebreaks :)

LazeMSS, 2021-01-06:
Bug found :) Navbar overflow problems

LazeMSS, 2021-01-09:
Closing - should be fixed in https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.2.4

closed 2021-01-09 by LazeMSS