LazeMSS/OctoPrint-UICustomizer #50: UI Costumizer UI gone bug with remove temp background

JRosello, 2021-01-02:
## Description
Describe your problem?
Marked the option to hide temp background
What is the problem?
All UI is gone except top and low bars
Can you recreate it?
Yes
Did you try disabling plugins?
If you disables de plugin normal UI works, as soon you reenable the plugin UI is gone 
Did you remember to update the subject?**
<hr>

**Plugins installed**
- FileManager[filemanager] v0.1.4
- Fullscreen Plugin[fullscreen] v0.0.6
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- Navbar Temperature Plugin[navbartemp] v0.14
- Octolapse[octolapse] v0.4.1
- PrintJobHistory[PrintJobHistory] v1.10.0
**- **UI Customizer[uicustomizer] v0.1.2.0****

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: false
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:84.0) Gecko/20100101 Firefox/84.0

LazeMSS, 2021-01-02:
This is a duplicate of another bug I have - this is really strange - have you fixed it? Or is there anyway I can have a look at your screen/teamviewer etc. to try and look into what happening?

LazeMSS, 2021-01-02:
Or could you post the output of your developer console?

seebaer1976, 2021-01-02:
have the same problem.
with update to 0.1.2.0 - Fixed 0.1.1.9 release, everything is gone except for the header and footer.
see image:

[image]

**Plugins installed**
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Cancel Objects[cancelobject] v0.4.4
- Dashboard[dashboard] v1.16.3
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.24.0
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoKlipper[klipper] v0.3.2
- PrettyGCode[prettygcode] v1.2.4
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.3
- Themeify[themeify] v1.2.2
- Tuya Smartplug[tuyasmartplug] v0.3.0
- **UI Customizer[uicustomizer] v0.1.2.0**

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: false

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 10.16; rv:84.0) Gecko/20100101 Firefox/84.0

LazeMSS, 2021-01-02:
Could you post output from developer console ?

LazeMSS, 2021-01-02:
I would really like to see if it can be recreated after installing the new version of my plugin - it should not be possible.

seebaer1976, 2021-01-02:
> Could you post output from developer console ?

stupid question. where is find again?

seebaer1976, 2021-01-02:
can you use them?

[octoprint.log](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5760551/octoprint.log)
[plugintimings.log](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5760552/plugintimings.log)

LazeMSS, 2021-01-02:
I wanted the output of: https://developer.mozilla.org/en-US/docs/Tools/Web_Console#Opening_the_Web_Console

LazeMSS, 2021-01-02:
https://developer.mozilla.org/en-US/docs/Tools/Browser_Console

seebaer1976, 2021-01-02:
here the output of the console:

[console-export-2021-1-2_18-18-16.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5760589/console-export-2021-1-2_18-18-16.txt)

JRosello, 2021-01-02:
Just for a clue.. if you clean the data on the plugin, re-enable and restart octorpint it works again

LazeMSS, 2021-01-02:
@JRosello but can you make it break again then?

JRosello, 2021-01-02:
Now Seems fine...
I think the problem came when you install the Consolidated Tabs with **UI C**ustomizer plug-in when it is installed and adjusted to your own needs.
I tried uninstalling Consolidated Tabs, clear the **UIC** data, setup **UIC** and the reinstall Consolidated Tabs... now seems that is working.

LazeMSS, 2021-01-02:
Ahh but you did not list Consolidated tabs in the report?

JRosello, 2021-01-02:
Sorry my mistake, when UIC failed Consolidated tabs wasn't installed yet

LazeMSS, 2021-01-02:
Okay - I have a new version out - Im still unable to recreate it but now I can check for the error at least

JRosello, 2021-01-02:
Great thanks!!

LazeMSS, 2021-01-02:
Please try again :)

JRosello, 2021-01-03:
Well tried and now seems to work... I'll reinstall Consolidated Tabs and see how it goes... If NO news Good news

Thanks for your work!!

LazeMSS, 2021-01-03:
I will close for now

closed 2021-01-03 by LazeMSS