LazeMSS/OctoPrint-UICustomizer #51: Compatibility issue with SimpleEmergencyStop

Sebclem, 2021-01-04:
Hi,
I'm the developer of [SimpleEmergencyStop](https://github.com/Sebclem/OctoPrint-SimpleEmergencyStop). It seem that we have a compatibility issue.

The navbar button seam to be unaligned when your plugin is installed:
Normal disposition: 
[image]
With your plugin installed:
[image]

I'm opening an issue here to know if I need to do something to make my plugin compatible with UiCustomizer or if the cards is all in your hands.

Let me know if you need more informations.
Opened issue on my repo : https://github.com/Sebclem/OctoPrint-SimpleEmergencyStop/issues/18

LazeMSS, 2021-01-04:
@Sebclem i will look - btw. to a quick look at your plugin - don't think you need to include font awesome 5 anymore :)

Sebclem, 2021-01-04:
Oh thank ! I see the update PR ! Perfect !

LazeMSS, 2021-01-04:
I think the error is on your side - you are including a "link" directly in the header - will update in your issue and close this

closed 2021-01-04 by LazeMSS