LazeMSS/OctoPrint-UICustomizer #53: Compact "Navbar Temperature Plugin" - ":" (colon) not deleted - ENHANCEMENT

LMS0815, 2021-01-04:
## Description
The Compact "Navbar Temperature Plugin" is displaying an unwanted  ":" (colon).
[image]

The colon should be hidden.
The Navbar Temperature Plugin has an issue displaying the PI internal temperature.
Therefor I had to create a Custom Command: `vcgencmd measure_temp | cut -d"=" -f2 | sed "s/'/\&#176;/g"`

[image]

For some reason this adds a SPACE prior to the colon, which seems to create problems cleaning the line.
[image]

[image]
<hr>

**Plugins installed**
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Autologin Configuration[autologin_config] v0.2.0
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Firmware Updater[firmwareupdater] v1.7.6
- hue GCODE Commands[huecommands] v0.2.4
- M117PopUp[M117PopUp] v0.6.2
- M73 Progress Plugin[m73progress] v0.2.1
- Navbar Temperature Plugin[navbartemp] v0.14
- Plotly Temp Graph[plotlytempgraph] v0.0.4rc4
- PrettyGCode[prettygcode] v1.2.4
- Resource Monitor[resource_monitor] v0.2.7
- Smart Preheat[smartpreheat] v0.0.5
- Terminal Messaging[terminalmessaging] v0.1.0
- **UI Customizer[uicustomizer] v0.1.2.2**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36

LazeMSS, 2021-01-04:
Try using the "Display internal temp" option - it does the same as your are doing there and I think it will work better.

Im working with the authors of navbar temp and improving it - the functions in my code are just hacks. I will move them into the navbar plugin eventually.

Closing this issue

closed 2021-01-04 by LazeMSS

LMS0815, 2021-01-06:
The _Navbar Temperature Plugin_ has a **known issue** displaying the PI internal temperature if you are using most current setup. If I enable internal temperature, the plugin crashes. Therefor I am using the workaround described above.

Software versions:
OctoPrint 1.5.2
Python 3.7.3
OctoPi 0.18.0

Updating to **UI Customizer[uicustomizer] v0.1.2.3** did not solve the issue.
[image]

LMS0815, 2021-01-06:
Updating to **UI Customizer[uicustomizer] v0.1.2.3** and changing _Navbar Temperature Plugin_ settings (see below) solved the issue.

[image]

LazeMSS, 2021-01-06:
I will hopefully soon have time to fix the navbar temp stuff :)

LMS0815, 2021-01-06:
**THX**. Its not urgent anyway.
Talking about temperature; [OctoPrint-PlotlyTempGraph](https://github.com/jneilliii/OctoPrint-PlotlyTempGraph) is very interesting plugin. It could be a source of temperatures for your own _Navbar_ widget.

LazeMSS, 2021-01-06:
Is that bug in the OctoPi 0.18.0 - on a RPI4?

LMS0815, 2021-01-07:
I can see the issue is on on my PI3 B+ ( OctoPrint 1.5.2, Python 3.7.3, OctoPi 0.18.0).
I remember reading that is caused by OctoPi 0.18.0 (64 bit) and the new underlying OS.