LazeMSS/OctoPrint-UICustomizer #54: Critical Failure on Saving

jneilliii, 2021-01-04:
## Description
[image]

```
packed_core.js?8be92d93:795 Error calling onSettingsHidden on view model UICustomizerViewModel : TypeError: Cannot read property 'destroy' of undefined
    at UICustomizerViewModel.self.onSettingsHidden (http://192.168.0.120/static/webassets/packed_plugins.js?d3730ba7:23926:73)
    at callViewModelIf (http://192.168.0.120/static/webassets/packed_core.js?8be92d93:794:126)
    at http://192.168.0.120/static/webassets/packed_core.js?8be92d93:788:138
    at Pn (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:94753)
    at Function.<anonymous> (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:105184)
    at callViewModelsIf (http://192.168.0.120/static/webassets/packed_core.js?8be92d93:788:95)
    at callViewModels (http://192.168.0.120/static/webassets/packed_core.js?8be92d93:787:56)
    at HTMLDivElement.<anonymous> (http://192.168.0.120/static/webassets/packed_core.js?8be92d93:242:848)
    at HTMLDivElement.dispatch (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:43090)
    at HTMLDivElement.v.handle (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:41074)
    at Object.trigger (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:71513)
    at HTMLDivElement.<anonymous> (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:72108)
    at Function.each (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:2976)
    at S.fn.init.each (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:1454)
    at S.fn.init.trigger (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:72084)
    at Modal.hideModal (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:560:22)
    at HTMLDivElement.<anonymous> (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:559:605)
    at HTMLDivElement.i (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:40056)
    at HTMLDivElement.dispatch (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:43090)
    at HTMLDivElement.v.handle (http://192.168.0.120/static/webassets/packed_libs.js?ab739628:1:41074)
    at HTMLDivElement.sentryWrapped (http://192.168.0.120/static/webassets/packed_core.js?8be92d93:457:45795)
```
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.2.2**
- Terminal Commands Extended[terminalcommandsextended] v0.1.7rc2
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.4rc1
- Cura Thumbnails[UltimakerFormatPackage] v0.2.1
- TP-Link Smartplug[tplinksmartplug] v0.9.26
- Tasmota[tasmota] v0.8.17
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- MQTT Subscribe[mqttsubscribe] v0.1.6
- M117 Speech Synthesis[M117SpeechSynthesis] v0.5.2
- GPIO Shutdown[GPIOShutdown] v1.0.3
- Custom Background[custombackground] v1.0.0
- Active Filters Extended[active_filters_extended] v0.1.0
- Google Drive Backup[googledrivebackup] v0.0.3

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- disableTermInactive: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36

peterwup, 2021-01-04:
Same problem for me: 

## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- **UI Customizer[uicustomizer] v0.1.2.2**
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.3
- Prusa ETA override Plugin[PrusaETAOverride] v0.1.1
- Telegram Notifications[telegram] v1.6.2
- Prusa Leveling Guide[PrusaLevelingGuide] v1.0.12
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrettyGCode[prettygcode] v1.2.4
- Fullscreen Plugin[fullscreen] v0.0.6
- Filament Manager[filamentmanager] v1.6.3
- Octolapse[octolapse] v0.4.1
- Bed Visualizer[bedlevelvisualizer] v1.0.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: true
- mainTabsCustomize: false
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36

rebeltaz, 2021-01-05:
Same issue here after upgrade to latest release.

jneilliii, 2021-01-05:
I had to revert back to version 0.1.1.8 in order for the plugin to not error onSettingsShown/onSettingsHidden in the console by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking install.

https://github.com/LazeMSS/OctoPrint-UICustomizer/archive/0.1.1.8.zip

LazeMSS, 2021-01-05:
@jneilliii  tried deleting data from my plugin? I would really love a teamviewer preview of whats happening.

LazeMSS, 2021-01-05:
Bug has been found - can be solved temporarily by deleting the settings from the plugin.

closed 2021-01-05 by LazeMSS