LazeMSS/OctoPrint-UICustomizer #55: SoC temperature in Nav Bar Temp is cut at the left

Ajimaru, 2021-01-04:
Describe your problem?
SoC temperature in Nav Bar Temp is cut at the left

What is the problem?
[image]

Can you recreate it?
yes

Did you try disabling plugins?
Yes I have disabled the plugin Navbar Temperature Plugin (0.14)
Result is that no temperautre are displayed in the nav bar :-)

Did you remember to update the subject?**
<hr>

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- **UI Customizer[uicustomizer] v0.1.2.2**
- Themeify[themeify] v1.2.2
- Terminal Messaging[terminalmessaging] v0.1.0
- STL Viewer[stlviewer] v0.4.2
- Sticky Pad[stickypad] v0.1.4
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.3
- PrintJobHistory[PrintJobHistory] v1.10.0
- Preheat Button[preheat] v0.5.1
- WS281x LED Status[ws281x_led_status] v0.6.0
- OctoPrint-WiFiStatus[wifistatus] v1.4.0
- Cura Thumbnails[UltimakerFormatPackage] v0.2.1
- Touchtest Plugin[touchtest] v0.4.1
- Tempsgraph Plugin[tempsgraph] v0.3.6
- Telegram Notifications[telegram] v1.6.2
- Tasmota[tasmota] v0.8.17
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- Resource Monitor[resource_monitor] v0.2.7
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrettyGCode[prettygcode] v1.2.4
- Navbar Temperature Plugin[navbartemp] v0.14
- MultiCam[multicam] v0.2.8
- MQTT Subscribe[mqttsubscribe] v0.1.6
- MQTT[mqtt] v0.8.7
- GPIO Shutdown[GPIOShutdown] v1.0.3
- GcodeEditor[GcodeEditor] v0.2.10
- Firmware Updater[firmwareupdater] v1.7.6
- Filament Manager[filamentmanager] v1.6.3
- Enclosure Plugin[enclosure] v4.13.1
- DisplayProgress[displayprogress] v0.1.3
- Dashboard[dashboard] v1.16.3
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- Marlin EEPROM Editor[eeprom_marlin] v3.0.1
- Macro[macro] v0.3.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.24.0
- Backup Scheduler[backupscheduler] v0.0.4

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- disableTermInactive: false
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: false
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0

LazeMSS, 2021-01-05:
The stuff I made is a bit of hack. I will probably make a better solution ASAP

LazeMSS, 2021-01-05:
I might have fixed it in next release :)

closed 2021-01-05 by LazeMSS