LazeMSS/OctoPrint-UICustomizer #58: Error: "Critical failure saving UI Customizer settings - not saved!"

TheKrush, 2021-01-05:
[image]

This is happening since the last 2 updates.

TheKrush, 2021-01-05:
Actually just saw patch notes, this might be a duplicate of #54 since the new patch isn't out on my octoprint yet.

Ajimaru, 2021-01-06:
Unfortunetly the issue is still occruring in conjunction with the Plugin "Macro" Version 0.3.0

[image]

The version of UI Customizer is 0.1.2.3
Here is the consaole log
[browser console log.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5776477/browser.console.log.txt)

LazeMSS, 2021-01-06:
@Ajimaru  could you post your config.yaml part for my plugin?

LazeMSS, 2021-01-06:
Its not related to the plugin - I just tested it

Ajimaru, 2021-01-06:
uicustomizer:
        addWebCamZoom: false
        compressTempControls: false
        fixedFooter: false
        gcodeFullWidth: true
        hideGraphBackground: false
        hideMainCam: true
        mainTabs:
        -   - tab_plugin_consolidate_temp_control_link
            - true
            - false
            - fas fa-gamepad
            - true
            - '#000000'
        -   - tab_plugin_dashboard_link
            - true
            - false
            - fas fa-tachometer-alt
            - true
            - '#000000'
        -   - tab_plugin_prettygcode_link
            - true
            - Gcode
            - fab fa-codepen
            - true
            - '#000000'
        -   - term_link
            - true
            - Terminal
            - fas fa-terminal
            - true
            - '#000000'
        -   - tab_plugin_resource_monitor_link
            - true
            - false
            - fas fa-memory
            - true
            - '#000000'
        -   - tab_plugin_PrintJobHistory_link
            - true
            - false
            - fas fa-history
            - true
            - '#000000'
        -   - timelapse_link
            - true
            - false
            - fas fa-film
            - true
            - '#000000'
        -   - gcode_link
            - true
            - Gcode Viewer
            - fab fa-codepen
            - true
            - '#000000'
        -   - tab_plugin_stlviewer_link
            - true
            - false
            - fab fa-codepen
            - true
            - '#000000'
        -   - tab_plugin_tasmota_link
            - true
            - false
            - fas fa-power-off
            - true
            - '#000000'
        -   - tab_plugin_enclosure_link
            - true
            - false
            - fas fa-window-maximize
            - true
            - '#000000'
        -   - tab_plugin_eeprom_marlin_link
            - true
            - false
            - fas fa-microchip
            - true
            - '#000000'
        rows:
        -   _00#connection_wrapper: true
            _01#state_wrapper: true
            _02#sidebar_plugin_filamentmanager_wrapper: true
            _03#files_wrapper: true
            _04#UICGcodeVWidget: false
        -   _00div.UICmainTabs: true
        -   _00#sidebar_plugin_macro_wrapper: true
            _01#sidebar_plugin_touchtest_wrapper: true
            _02#sidebar_plugin_firmware_check_wrapper: true
            _03#sidebar_plugin_action_command_notification_wrapper: true
            _04#UICWebCamWidget: true
            _05#sidebar_plugin_tasmota_wrapper: true
        topIconSort:
        - navbar_plugin_navbartemp
        - navbar_plugin_DisplayLayerProgress
        - navbar_plugin_simpleemergencystop
        - navbar_plugin_ws281x_led_status
        - navbar_plugin_tasmota
        - navbar_plugin_stickypad
        - navbar_plugin_wifistatus
        - navbar_plugin_firmwareupdater
        - navbar_plugin_enclosure_1
        - navbar_plugin_enclosure_2
        - navbar_plugin_pi_support
        - navbar_plugin_prettygcode
        - navbar_plugin_PrintJobHistory
        - navbar_plugin_action_command_prompt
        - navbar_plugin_logging_seriallog
        - navbar_plugin_logging_plugintimingslog
        widths:
        - '3'
        - '6'
        - '3'

LazeMSS, 2021-01-06:
@Ajimaru come online on discord so we maybe can test better.
Dont see anything that should cause that error in your config.

Ajimaru, 2021-01-07:
@LazeMSS yes discord is fine.

Today I got a crash after uninstalling the abandoned Cost plugin, browser console output is attached,
[exeption after delete cost plugin.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5783040/exeption.after.delete.cost.plugin.txt)

Ajimaru, 2021-01-07:
crash again, I have attached the console log and the entire content of the links you mentioned.

[crash log.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5783938/crash.log.txt)
[packed_plugins_js_02da810b_37769_33.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5783939/packed_plugins_js_02da810b_37769_33.txt)
[packed_plugins_js_02da810b_37775_27.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5783944/packed_plugins_js_02da810b_37775_27.txt)

LazeMSS, 2021-01-08:
Did you install any new plugins or did you just change settings.

Ajimaru, 2021-01-08:
I think i have installed a new plugin.

LazeMSS, 2021-01-09:
Latest version has more debug in the console if it happens again

Ajimaru, 2021-01-13:
I have opend Octoprint Config to change a folder, crash occured after pressing  the save button.

[console-export-2021-1-13_12-1-56.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5808188/console-export-2021-1-13_12-1-56.txt)
[http192.168.1.1095000staticwebassetspacked_plugins.jsa5be22093789233.txt](https://github.com/LazeMSS/OctoPrint-UICustomizer/files/5808189/http192.168.1.1095000staticwebassetspacked_plugins.jsa5be22093789233.txt)

**Plugins installed**
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- Backup Scheduler[backupscheduler] v0.0.4
- Consolidate Temp Control[consolidate_temp_control] v0.1.8
- Cost Estimation[costestimation] v3.1.0
- Cura Thumbnails[UltimakerFormatPackage] v0.2.1
- Dashboard[dashboard] v1.17.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.24.0
- DisplayProgress[displayprogress] v0.1.3
- Enclosure Plugin[enclosure] v4.13.1
- Filament Manager[filamentmanager] v1.6.3
- Firmware Updater[firmwareupdater] v1.7.6
- GcodeEditor[GcodeEditor] v0.2.10
- GPIO Shutdown[GPIOShutdown] v1.0.3
- Macro[macro] v0.3.0
- Marlin EEPROM Editor[eeprom_marlin] v3.0.1
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- MQTT[mqtt] v0.8.7
- MQTT Subscribe[mqttsubscribe] v0.1.6
- MultiCam[multicam] v0.2.8
- Navbar Temperature Plugin[navbartemp] v0.14
- OctoPrint-WiFiStatus[wifistatus] v1.4.0
- Preheat Button[preheat] v0.5.1
- PrettyGCode[prettygcode] v1.2.4
- PrintJobHistory[PrintJobHistory] v1.10.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.3
- Resource Monitor[resource_monitor] v0.2.8
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- Sticky Pad[stickypad] v0.1.4
- STL Viewer[stlviewer] v0.4.2
- SwapXY[swapxy] v0.3.0
- Tasmota[tasmota] v0.8.17
- Telegram Notifications[telegram] v1.6.2
- Tempsgraph Plugin[tempsgraph] v0.3.6
- Terminal Messaging[terminalmessaging] v0.1.0
- Themeify[themeify] v1.2.2
- Touchtest Plugin[touchtest] v0.4.1
- **UI Customizer[uicustomizer] v0.1.2.4**
- WS281x LED Status[ws281x_led_status] v0.6.0
- Z Probe Offset[z_probe_offset] v0.5

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: false
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: false
- hideMainCam: true
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0

[image]

Ajimaru, 2021-01-13:
I have disabled Ui-Customizer due to incompatibility with TouchUi

LazeMSS, 2021-01-13:
Might have found a bug that could cause it

LazeMSS, 2021-01-18:
Closing - should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.2.5
If anyone disagrees please reopen this issue :)

closed 2021-01-18 by LazeMSS