LazeMSS/OctoPrint-UICustomizer #59: SoC temp still cut on the left

Ajimaru, 2021-01-05:
Describe your problem?
SoC temp is still cut on the left with lates Ui Customizer update
[image]

What is the problem?
Two digits left of the . are cut off.

Can you recreate it?
yes

Did you try disabling plugins?
yes, also delete the Ui Customizer plugin data.
[image]

Did you remember to update the subject?

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- **UI Customizer[uicustomizer] v0.1.2.3**
- Themeify[themeify] v1.2.2
- Terminal Messaging[terminalmessaging] v0.1.0
- STL Viewer[stlviewer] v0.4.2
- Sticky Pad[stickypad] v0.1.4
- PrusaSlicer Thumbnails[prusaslicerthumbnails] v0.1.3
- PrintJobHistory[PrintJobHistory] v1.10.0
- Preheat Button[preheat] v0.5.1
- WS281x LED Status[ws281x_led_status] v0.6.0
- OctoPrint-WiFiStatus[wifistatus] v1.4.0
- Cura Thumbnails[UltimakerFormatPackage] v0.2.1
- Touchtest Plugin[touchtest] v0.4.1
- Tempsgraph Plugin[tempsgraph] v0.3.6
- Telegram Notifications[telegram] v1.6.2
- Tasmota[tasmota] v0.8.17
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- Resource Monitor[resource_monitor] v0.2.7
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PrettyGCode[prettygcode] v1.2.4
- Navbar Temperature Plugin[navbartemp] v0.14
- MultiCam[multicam] v0.2.8
- MQTT Subscribe[mqttsubscribe] v0.1.6
- MQTT[mqtt] v0.8.7
- GPIO Shutdown[GPIOShutdown] v1.0.3
- GcodeEditor[GcodeEditor] v0.2.10
- Firmware Updater[firmwareupdater] v1.7.6
- Filament Manager[filamentmanager] v1.6.3
- Enclosure Plugin[enclosure] v4.13.1
- DisplayProgress[displayprogress] v0.1.3
- Dashboard[dashboard] v1.16.3
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- Marlin EEPROM Editor[eeprom_marlin] v3.0.1
- Macro[macro] v0.3.0
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.24.0
- Backup Scheduler[backupscheduler] v0.0.4

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:84.0) Gecko/20100101 Firefox/84.0

LazeMSS, 2021-01-05:
Dont make a new issue for an existing one - just comment on the closed on ... Please.

You are not using standard settings - its says "Pi:.." - try chaning the "SoC name to display" to "SoC" and check all 3 checkboxes. That will fix it.

closed 2021-01-05 by LazeMSS

Ajimaru, 2021-01-05:
Well the other issue was closed, sorry.
I have changed the name to SoC, I was not aware that this has to be done, thanks.

P.S. good work 👍

LazeMSS, 2021-01-05:
Okay thanks. I was a bit stressed sorry if I sounded rude

thelastWallE, 2021-09-06:
> 
> 
> Dont make a new issue for an existing one - just comment on the closed on ... Please.
> 
> You are not using standard settings - its says "Pi:.." - try chaning the "SoC name to display" to "SoC" and check all 3 checkboxes. That will fix it.

If i had not seen this here i would have opened a new issue for this. This needs definitely an info somewhere.

LazeMSS, 2021-09-06:
I will update the docs - and maybe check the settings to navbar.

I was actually planning to remove support for the navbar temp stuff.