LazeMSS/OctoPrint-UICustomizer #60: This plugin screws up TouchUI

mdaneman, 2021-01-06:
When this plugin is enabled, TouchUI scaling on mobile devices is all screwed up.  It would be great if this could be fixed.

LazeMSS, 2021-01-06:
What is problem exactly?
There is a problem together with Touch UI plugin?

mdaneman, 2021-01-06:
I guess I was assuming that TouchUI will still take over on mobile devices.  I turned-off TouchUI and see that this plugn does work on a phone by scrolling through pages, but the TouchUI tab-based interface is definitely more usable on a small screen.  Is there a way to have this plugin give up control to TouchUI for small screens?  I do like UICusomizer for large screens though.

LazeMSS, 2021-01-06:
Yes the TouchUI is a complete UI take over - the only solution i guess is to make my UI changes to be hidden when using the #touch part of the UI. I will look into it later, I haven't researched how TouchUI worked.

cereyanli, 2021-02-18:
Any progress on this enhancement so far

LazeMSS, 2021-02-18:
Will look asap - only so many free hours 🙈

LazeMSS, 2021-02-21:
I have fixed in the next - but i requires you load the UI using #touch added to the URL when loading on the touch screens. This is best fix for all I think.

LazeMSS, 2021-02-21:
Fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.2

Closing - please reopen if not working/fixed as expected

closed 2021-02-21 by LazeMSS