LazeMSS/OctoPrint-UICustomizer #62: Terminal Screen UI Stuttering

zer0ish, 2021-01-07:
Since the last update, when I go into the  terminal window the whole UI start to stutter. Like its in a constant mode of changing sizes. I added a small clip to show what it's doing.

https://user-images.githubusercontent.com/29928635/103926605-24ccc880-50e7-11eb-8965-9f813cfc8864.mp4

LazeMSS, 2021-01-07:
What plugins are you using to put the webcam in terminal window?

zer0ish, 2021-01-07:
It's part of the UI Customizer plugin. No other plugin needed to put the webcam there.
[image]

LazeMSS, 2021-01-07:
Ahh sorry and it happend after last update?

zer0ish, 2021-01-07:
I believe so, I dont spend too much time in the terminal window area, so can't be 100% sure. I just started trying to fine tune my calibrations and now I'm in it a bit more so I noticed this behavior. Most likely something to do with the text in the terminal window the more and faster the text populates the more the jittering happens.

zer0ish, 2021-01-07:
New info.

When on a 1080p monitor I dont get this issue. Full screen 1440p, no issues, when I split my 1440p monitor, which I do most of the time when doing 3d printing and browsing general work, the problem happens. So the video I put up, was half my 1440p monitor.

Not sure why this is happening.

LazeMSS, 2021-01-08:
Thanks I will look into it ASAP

LazeMSS, 2021-01-08:
@zer0ish  do you have the stutter without my plugin?

zer0ish, 2021-01-08:
Ill check once my print is done, Ill disable your plugin and test. Or should I uninstall it completely to test?

LazeMSS, 2021-01-08:
Just disable in plugin manager - i tried running in 2560 x 1440 pixels and same layout - could not recreate the problem on my part

zer0ish, 2021-01-09:
Ok, I disabled the plugin, restarted my octopi, problem is gone. Terminal window doesn't jitter. Turned plugin on, restarted. Jitter is back.

LazeMSS, 2021-01-09:
Can you see if it happens on smaller/larger resolutions - just by resizing the browser a couple of pixels etc.

zer0ish, 2021-01-09:
I think the problem isn't with your plugin. I disabled Themeify's theme that made it dark mode and the issue goes away. 
You can close this as the problem is with themeify.
Any chance you can add a dark mode setting for UI Customizer so I don't have to use themeify?

LazeMSS, 2021-01-09:
Yes I am looking into making my own themes also - but only so many hours of my free time

closed 2021-01-09 by LazeMSS