LazeMSS/OctoPrint-UICustomizer #64: Main Tab Icon Colors

brombomb, 2021-01-10:
## Description
The main tab icon colors don't match the rest of the site/Themeify.   I have the themeify icons turned off and still can't override it.

Notice the Webcam which is an accordion header is colored light, but the main tabs are all dark.
[image]

Themeify CSS
[image]

Main Tab (UICustomizezer CSS)
[image]

<hr>

**Plugins installed**
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Backup Scheduler[backupscheduler] v0.0.4
- Cost Estimation[costestimation] v3.1.0
- Filament Manager[filamentmanager] v1.6.3
- FileManager[filemanager] v0.1.4
- Fullscreen Plugin[fullscreen] v0.0.6
- Google Drive Backup[googledrivebackup] v0.0.3
- M117PopUp[M117PopUp] v0.6.2
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- MultipleUpload Plugin[MultipleUpload] v1.1.0
- Octoprint-Display-ETA[display_eta] v1.1.2
- OctoPrint-WiFiStatus[wifistatus] v1.4.0
- PrettyGCode[prettygcode] v1.2.4
- Resource Monitor[resource_monitor] v0.2.7
- Terminal Commands Extended[terminalcommandsextended] v0.1.6
- Themeify[themeify] v1.2.2
- **UI Customizer[uicustomizer] v0.1.2.4**
- Upload Anything[uploadanything] v1.0.1
- WebcamTab[webcamtab] v0.2.0

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3
- OctoPi 0.18.0

**Browser**
- Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.88 Safari/537.36

LazeMSS, 2021-01-10:
Ahh will look

closed 2021-01-11 by LazeMSS

brombomb, 2021-01-11:
Sorry I probably sent you on a goose chase the wrong way.  The webcam color to me is nicly colored.  I'd like those accordion colors to persist into the Main Tabs (Controls, Temp, Layers, etc), but they have an inline css of `rgb(0,0,0)` which is really hard to override.  If the inline style weren't there it's be the correct color as defined by Themeify ```.themeify.discorded a {
    color: #99aab5;
}```

LazeMSS, 2021-01-12:
Ahh sorry

LazeMSS, 2021-01-12:
I will add an option to set the color "none"