LazeMSS/OctoPrint-UICustomizer #66: Enhacement Request - Temp Graph Colors

RAMilewski, 2021-01-10:
I frequently run Octoprint with the browser plugin Dark Reader which adds a dark mode to most web pages.  It would be helpful to be able to specify the colors of the lines in the temperature graph.

LazeMSS, 2021-01-11:
I would recommend themeify and one of its dark themes.

RAMilewski, 2021-01-13:
I used to use Themeify, but it didn't play well with some other plugins.  I just reinstalled it and tried again... No joy!   It messes up the Navbar Temp plugin and the dark themes don't measurably improve the temp graph. 

I did try the Tempsgraph plugin and that seems to help some, although the colors are still not configurable.    Probably the best solution for now.

closed 2021-01-13 by RAMilewski