LazeMSS/OctoPrint-UICustomizer #67: Double background

Lecl0ne, 2021-01-12:
It looks like the background is kind of double, with one rounded and the other not.
Did you try disabling plugins? Yes, it solves the problem
<hr>

**Plugins installed**
- ProgressBasedOntime Plugin[ProgressBasedOnTime] v0.1.6
- Friendly Neighborhood Beeper[FriendlyBeeper] v0.4.0
- Marlin GCode Documentation[marlingcodedocumentation] v0.11.0
- Fullscreen Plugin[fullscreen] v0.0.6
- Auto Terminal Input[AutoTerminalInput] v0.1.0
- Themeify[themeify] v1.2.2
- Autoscroll[autoscroll] v0.0.3
- Terminal Messaging[terminalmessaging] v0.1.0
- **UI Customizer[uicustomizer] v0.1.2.4**
- Autoselect Plugin[autoselect] v0.3.0
- Autologin Configuration[autologin_config] v0.2.0
- M117NavBar[M117NavBar] v0.1.1
- Printoid Plugin[printoid] v1.0.2
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- SetAndWait[setandwait] v0.0.1
- Progress Title[progress_title] v0.1.1
- Detailed Progress[detailedprogress] v0.2.6
- GCODE System Commands[gcodesystemcommands] v1.0.1
- Cura Thumbnails[UltimakerFormatPackage] v0.2.1
- Preheat Button[preheat] v0.5.1
- Multi Line Terminal[multilineterminal] v0.1.5
- SpoolManager Plugin[SpoolManager] v1.2.0
- Cost Estimation[costestimation] v3.1.0
- Tempsgraph Plugin[tempsgraph] v0.3.6
- Active Filters Extended[active_filters_extended] v0.1.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Octolapse[octolapse] v0.4.1

**UI Customizer settings**
- addWebCamZoom: false
- centerTopIcons: true
- compactMenu: true
- compressTempControls: false
- disableTermInactive: true
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: true
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36 Edg/87.0.664.75

[image]

LazeMSS, 2021-01-12:
Its related to a theme I think - what theme are you using?

Lecl0ne, 2021-01-12:
Discoranged. If I disable this theme, it solves the problem. If i turn it back on, it's here again...

LazeMSS, 2021-01-12:
I will look

LazeMSS, 2021-01-18:
Fixed in next release

LazeMSS, 2021-01-18:
Closing - should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.2.5
If anyone disagrees please reopen this issue :)

closed 2021-01-18 by LazeMSS

Lecl0ne, 2021-01-19:
Double background solved. But the shadow is kind of FAT, comparing to the normal one.
[image]

LazeMSS, 2021-01-19:
Ahh will look