LazeMSS/OctoPrint-UICustomizer #68: Expanding secondary camera with multicam plugin shows wrong camera

gonzonia, 2021-01-12:
## Description
**ENTER DESCRIPTION HERE
Describe your problem? With Multicam plugin, when secondary camera is selected, if the camera window is expanded, it shows the primary camera. 
What is the problem? See above
Can you recreate it? Yes
Did you try disabling plugins? No
<hr>

**Plugins installed**
- Access Anywhere - The Spaghetti Detective[thespaghettidetective] v1.4.7
- Arc-Welder[arc_welder] v1.0.0+u.bb71e8f
- Exclude Region[excluderegion] v0.3.0
- MultiCam[multicam] v0.2.8
- OctoPod Plugin[octopod] v0.2.9
- OctoPrint-WiFiStatus[wifistatus] v1.4.0
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- Prusa Leveling Guide[PrusaLevelingGuide] v1.0.12
- TP-Link Smartplug[tplinksmartplug] v0.9.26
- **UI Customizer[uicustomizer] v0.1.2.4**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- fixedFooter: true
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.2
- Python 3.7.3
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Macintosh; Intel Mac OS X 11_1_0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36

LazeMSS, 2021-01-18:
I can't recreate this - i only have one cam but I have tried with multiple different public urls - what urls are you using?

gonzonia, 2021-01-18:
Thanks. To answer the question though in case it helps, I'm using a local IP address for one cam (running an old iPhone as a cam), and the second cam is directly connected to the Pi. 
http://<local ip>/live and http://<local IP for Pi>:8080/?action=stream

LazeMSS, 2021-01-18:
Closing - should be fixed in: https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.2.5
If anyone disagrees please reopen this issue :)

closed 2021-01-18 by LazeMSS