LazeMSS/OctoPrint-UICustomizer #71: icons

Ruissilva, 2021-01-22:
## Description
Hi.
These images show thu UI Customizer. In the computer its ok, but in the phone its wierd. Any ideia how i can fix this?

[image]
[image]

**Plugins installed**
- Access Anywhere - The Spaghetti Detective[thespaghettidetective] v1.4.7
- Autoscroll[autoscroll] v0.0.3
- Autoselect Plugin[autoselect] v0.3.0
- Backup Scheduler[backupscheduler] v0.0.4
- Bed Visualizer[bedlevelvisualizer] v1.0.0
- Cancel Objects[cancelobject] v0.4.4
- Cura Thumbnails[UltimakerFormatPackage] v0.2.1
- Custom Background[custombackground] v1.0.0
- Custom Control Editor[customControl] v0.2.2
- DisplayLayerProgress Plugin[DisplayLayerProgress] v1.24.0
- Enclosure Plugin[enclosure] v4.13.1
- Filament Manager[filamentmanager] v1.6.3
- Filament Sensors Revolutions[filamentrevolutions] v1.0.0
- Firmware Updater[firmwareupdater] v1.7.7
- Floating Navbar[floatingnavbar] v0.3.4
- GcodeEditor[GcodeEditor] v0.2.10
- HeaterTimeout[HeaterTimeout] v0.0.3
- Multi Colors[multi_colors] v1.0.17
- Navbar Temperature Plugin[navbartemp] v0.14
- Octolapse[octolapse] v0.4.1
- OctoLight[octolight] v0.1.0
- Preheat Button[preheat] v0.5.1
- PrintJobHistory[PrintJobHistory] v1.10.0
- Printoid Plugin[printoid] v1.0.2
- PrintTimeGenius Plugin[PrintTimeGenius] v2.2.7
- PSU Control[psucontrol] v0.1.10
- Resource Monitor[resource_monitor] v0.2.8
- Simple Emergency Stop[simpleemergencystop] v1.0.4
- SlicerSettingsTab[SlicerSettingsTab] v1.0.3
- Tab Order[taborder] v0.5.12
- Telegram Notifications[telegram] v1.6.2
- TemperatureFailsafe[TemperatureFailsafe] v0.2.3
- Themeify[themeify] v1.2.2
- TimeToFilament Plugin[TimeToFilament] v1.2.1
- **UI Customizer[uicustomizer] v0.1.2.5**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: false
- compactMenu: true
- compressTempControls: true
- disableTermInactive: true
- fixedFooter: false
- fixedHeader: true
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: false
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.3
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36

LazeMSS, 2021-01-22:
I'm working on a solution for this - but basically there is not room for the icons.

LazeMSS, 2021-02-04:
Fixed in latest release - its not perfect but working

Ruissilva, 2021-02-06:
Hello LazeMSS, thanks for the help.
I installed the latest update of the plugin, and im sorry to say nothing changed.

LazeMSS, 2021-02-06:
Can you post a screenshot? And what is the problem. The icons or the menu?

Ruissilva, 2021-02-06:
Ok, right now i have installed UI Customizer (0.1.2.8). 

[image]

[image]

One thing that i whould like to see in this plugin is the ability to chose which icons to show, for example, with some toggle buttons. I put an image below has a example. 

[image]

best of luck

LazeMSS, 2021-02-06:
Okay - i just tested - i can see the same problems.

Btw. good idea with an option to disable a top icon

LazeMSS, 2021-02-17:
Should be fixed in the latest release - please try again:
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.0

Reopen if not fixed - thanks.

closed 2021-02-17 by LazeMSS