LazeMSS/OctoPrint-UICustomizer #72: GCode Viewer NOT filling column

GigaTech, 2021-01-24:
Hi, brilliant plugin.

I have an issue, where the GCode Viewer is NOT filling the full width of the column

[image]

in the image, this is a small part in the center of the print bed.

Many Thanks.

LazeMSS, 2021-01-24:
It's strange I have had reports of this before. I will look ASAP

GigaTech, 2021-01-24:
my system info, if needed:

```
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 917016576
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin 1.1.8 (Github)
```

Thanks

LazeMSS, 2021-01-24:
Do you have the " Cancel Object Plugin" installed?
Have a look here: https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/43

LazeMSS, 2021-01-24:
Ahh can see from your screenshot now that you dont have the cancel object plugin - can you try and pres F12 and chrome and run this command:
$('#gcode_canvas').prop('class')

And post back what it returns

GigaTech, 2021-01-24:
Hi, no I do not have that plugin installed

[image]

it return ""

LazeMSS, 2021-01-24:
Are your sure you are running the latest version of my plugin?

GigaTech, 2021-01-24:
installed today from the plugin manager...

[image]

LazeMSS, 2021-01-24:
Can you try and push the "report a bug" button in my plugin and just dont create the bug report but just copy the text over into this issue. Im looking for which options you have enabled but there should nothing holding it back. 

Can you maybe also try disabling themeify?

GigaTech, 2021-01-24:
## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- Themeify[themeify] v1.2.2
- **UI Customizer[uicustomizer] v0.1.2.5**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- fixedFooter: false
- fixedHeader: false
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.3
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36

GigaTech, 2021-01-24:
themeify disabled... the same

[image]

## Description
**ENTER DESCRIPTION HERE
Describe your problem?
What is the problem?
Can you recreate it?
Did you try disabling plugins?
Did you remember to update the subject?**
<hr>

**Plugins installed**
- Z Probe Offset[z_probe_offset] v0.5
- **UI Customizer[uicustomizer] v0.1.2.5**

**UI Customizer settings**
- addWebCamZoom: true
- centerTopIcons: true
- compactMenu: true
- compressTempControls: true
- disableTermInactive: false
- fixedFooter: false
- fixedHeader: false
- fluidLayout: true
- gcodeFullWidth: false
- hideGraphBackground: true
- hideMainCam: false
- mainTabsCustomize: true
- navbarplugintempfix: true
- responsiveMode: true

**Software versions**
- OctoPrint 1.5.3
- Python 2.7.16
- OctoPi 0.17.0

**Browser**
- Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.141 Safari/537.36

LazeMSS, 2021-01-25:
You need to enable the feature:
[image]

Closing

GigaTech, 2021-01-25:
Hi, Thanks for the info.

I left this box unticked as it creates a very long page and jumps to the top of the box then the mouse goes into the preview box, it is unusable in this state.

below is the full window screenshot

[image]

LazeMSS, 2021-01-25:
Well thats the behaviour of the current way of making it full size - the display has to be a square - it might look into some hacks but there are a lot of stuff going onside in the gcode viewer

GigaTech, 2021-01-25:
OK Thanks.

I was assuming that it would look like your images on your front page, such are the 'floating webcam' image / Temperature page

...I just changed the width and it looks OK (with the checkbox "Gcode full-size" unchecked), but then obviously there is more to it than this.

`<canvas id="gcode_canvas"...`

[image]

Thanks for your time, I am looking forward to an update.
Many Thanks

LazeMSS, 2021-01-25:
Well it looks like that on certain screen resolution due to the aspect ratios

The problem is aspect ratios - the screen get to tall - and some users like that and other don't - so I added it and option and it works like this for the moment.

LazeMSS, 2021-01-25:
I will close for now - I have a better feature planned... :)

closed 2021-01-25 by LazeMSS