LazeMSS/OctoPrint-UICustomizer #74: UI incorrect output

Homerist, 2021-02-03:
after 10 Minutes in print some screen failure. See Screenshot

After Reboot the Pi it works for about 10 minutes and then again.

[image]

LazeMSS, 2021-02-03:
You mean the gcode viewer?

Does it happen with all gcode files or just one certain?

Homerist, 2021-02-03:
with all, the same is at the temperature tab too

LazeMSS, 2021-02-03:
Have you tried disabling UI customizer and see if the problem is still there?

Homerist, 2021-02-03:
Yes, i tried it, after disabling UI customizer is all working normal

LazeMSS, 2021-02-04:
UI Customizer should not affect anything related to the temp graph and the bugs you have shown looks more like a rendering issue in a browser.

Homerist, 2021-02-04:
Please close this, i tried this on a other Pi and the same. But on a other browser it works. 

Sorry for that

closed 2021-02-04 by Homerist

LazeMSS, 2021-02-04:
Thanks for checking :)