LazeMSS/OctoPrint-UICustomizer #75: Consolidated Tabs Development Compatibility Issue

jneilliii, 2021-02-03:
I've run into an issue with my current development version of Consolidated Tabs that gets broken when used in conjunction with UI Customizer. Normally you panels that are created that allow resizing and dragging get collapsed for some reason and if you can let me know if there is anything I can do in my css to prevent this that would be greatly appreciated. 

* This is the plugin normally, you can see where the resizers are properly aligned in the bottom right corner of the panel.
[image]

* With UI Customizer enabled (resize navbar and use full width disabled in plugin's settings) that resizer gets squished to the top.
[image]

My current working development branch can be found here.
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/tree/develop/gridstack

and installed with this URL in Plugin Manager > Get More > ...from URL.
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/archive/develop/gridstack.zip

I'm sure it's something simple and I appreciate any help you can provide.

LazeMSS, 2021-02-04:
Can figure it out just right away but it seems like the div:
`<div class="grid-stack-item consolidated ui-draggable ui-resizable ui-resizable-autohide" gs-x="0" gs-y="0" gs-w="12" gs-h="25" gs-id="temp_link" draggable="true">
`
does not have a height. If you add a height it works. I think its due to the responsive boostrap that UI customizer includes.

I noticed if you remove the "position:absolute" from the div below it (class: grid-stack-item-content) - ie:

.grid-stack>.grid-stack-item>.grid-stack-item-content {
    margin: 0;
    /* position: absolute; */
    width: auto;
    overflow-x: hidden;
    overflow-y: auto;
}

it works fine then.

jneilliii, 2021-02-04:
Unless it's set in the css and done on load...

[image]

the tab completely vanishes and the positioning is all messed up.

[image]

jneilliii, 2021-02-04:
I've pushed that change to the same branch.

LazeMSS, 2021-02-04:
@jneilliii  does it fix it?

KL-XL, 2021-02-11:
Hi Both,

I don't know if this is related, but I keep getting this message from UI Customizer without any visible issues except the temperature table slightly overlapping the divider between the central and right column. Hope it is of help to resolve the issue as I love both plugins and the message is slightly annoying.

Thanks,
KL
[image]

LazeMSS, 2021-02-11:
I will remove in next release

LazeMSS, 2021-02-17:
Should be fixed in the latest release - please try again:
https://github.com/LazeMSS/OctoPrint-UICustomizer/releases/tag/0.1.5.0

Reopen if not fixed - thanks.

jneilliii, 2021-02-17:
This was a different issue then the one you fixed...and I can't re-open.

LazeMSS, 2021-02-20:
@jneilliii  i can't get anything to "work" no pencils are shown anywhere now.

I ran "OctoPrint.coreui.viewmodels.consolidatedtabsViewModel.toggleEditMode()" to enable it.

I general there will/are conflicts because we are "manipulating" the same stuff.

It seems to something with gridstack - i can fix it by removing
$('div.octoprint-container div.tabbable').addClass('UICmainTabs').wrap( '<div class="UICCol2"></div>');

But the problem is basically elsewhere because it happens as soon as the dom is altered. It seems to be related to the "dynamic" created stylesheet and then order of execution for our plugins.

So for instance this stylesheet is lost when the dom is updated.
```
.grid-stack-instance-8227 > .consolidated[gs-h="25"] {
    height: 625px;
}
```

notice 8227 is dynamic. The problem is that fixed height/top of everything is lost. You can try even use the dom inspector in their examples and move one outer container relative to another.

That stylesheet seems to dynamic created gridstack but it included deep in the page a not at the root of html which could be the cause. 
There are a couple of related issues on the gridstack ie https://github.com/gridstack/gridstack.js/issues/418 <- this is not 100% related but it goes into the way the set the styles can cause it to break...

LazeMSS, 2021-02-20:
@jneilliii  UPDATE - Just as I posted this i found it in the documentation:
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/blob/develop/gridstack/octoprint_consolidatedtabs/static/js/consolidatedtabs.js#L117
Just add this option:
`styleInHead : true,`

I'm closing it for now :)

closed 2021-02-20 by LazeMSS